modm-io/modm #583: Ethernet freertos_tcp on f439zi nucleo not working

kikass13, 2021-03-18:
Hello,
I will append my experience here, even though there is an issue for that already:
 * https://github.com/modm-io/modm/issues/447

### Preamble
I got @mikewolfram 's `lan8720a` phy driver and the corresponding example (using freertos_tcp) working with the f767zi nucleo board after changing some of the PHY pins hardcoded in the example he gave). 

### Getting There
After that, I tried to get it running on the f439zi but was unable to do so. After some hours of experimenting with the freertos network scheduler (RTOS_IP) and ethernet stack, i finally ended inside the event loop (controlled by freertos ip stack) of the `lan8720a` ([see here](https://github.com/modm-io/modm/blob/develop/ext/aws/modm_lan8720a.cpp#L421)) which should eventually handle arp requests coming from another laptop connected to the board via ethernet. 

### The Problem
The [event loop](https://github.com/modm-io/modm/blob/develop/ext/aws/modm_lan8720a.cpp#L421) handles events constantly (as expected), but every event is regarded as a NOP `modm::platform::eth::Event::None` and therefore ignored. Which means that the feeding function, responsible for setting proper events, is not working properly. 

As it turns out, the producer function is [this](https://github.com/modm-io/modm/blob/develop/ext/aws/modm_lan8720a.cpp#L623-L650) ISR 
```cpp
MODM_ISR(ETH)
{
	using modm::platform::eth;
	using modm::ethernet;

	BaseType_t xHigherPriorityTaskWoken = pdFALSE;
	EMAC::InterruptFlags_t irq = EMAC::getInterruptFlags();
	EMAC::acknowledgeInterrupt(irq);

	if (irq & (eth::InterruptFlags::Receive | eth::InterruptFlags::ReceiveBufferUnavailable)) {
		ethernet::isrEvent |= eth::Event::Receive;
		if (ethernet::emacTaskHandle) {
			vTaskNotifyGiveFromISR(ethernet::emacTaskHandle, &xHigherPriorityTaskWoken);
			portYIELD_FROM_ISR(xHigherPriorityTaskWoken);
		}
	}
	if (irq & (eth::InterruptFlags::Transmit)) {
		ethernet::isrEvent |= eth::Event::Transmit;
		if (ethernet::emacTaskHandle) {
			vTaskNotifyGiveFromISR(ethernet::emacTaskHandle, &xHigherPriorityTaskWoken);
			portYIELD_FROM_ISR(xHigherPriorityTaskWoken);
		}
	}

	if (EMAC::getInterruptFlags() & eth::InterruptFlags::AbnormalIrqSummary) {
		// not used yet
	}
}
```

My test have shown (led blinky, event chain debugging) that this specific ISR is not called properly. 
Is there something you guys can tell me about the differences between the interrupt handling (mainly ethernet-DMA interrupts) ?

### Other Stuff

* I tried checking registers for interrupts, dma and others to see if any flags have changed but it seems that these chips(f4 & f7) are equal in that matter
* we have an internal issue where this errata was evaluated, but we could not find anything in particular which could be problematic: https://www.st.com/resource/en/errata_sheet/dm00068628-stm32f427437-and-stm32f429439-line-limitations-stmicroelectronics.pdf

### End

Thank you for reading, maybe someone can help

salkinium, 2021-03-18:
Not a FreeRTOS expert, so I can only say some generic stuff:

- There's a minimum IRQ priority of I think 13 imposed by FreeRTOS, which uses lower priorities for SysTick and PendSV etc. Perhaps we're accidentally setting something wrong, or a lower prio interrupt is blocked by something higher?
- You can try `NVIC_SetPending(ETH_IRQn);` to test if the IRQ handler is even callable at all.
- There also a `ETH_WKUP` interrupt? Not sure what it does.
- Maybe there's some error state in the ethernet peripheral that's not being acknowledged, and therefore blocks new events? Maybe the `AbnormalIrqSummary` requires clearing some register in addition to the clearing flags.

kikass13, 2021-03-18:
> Not a FreeRTOS expert, so I can only say some generic stuff:

its not a freertos problem, so your answers are helpful.
Im gonna try some of the things you mentioned and get back as soon as possible (could take some days)

mikewolfram, 2021-03-18:
The ethernet implementation is an adaption of the code provided in FreeRTOS+TCP for STM32, which uses the STM32 HAL. There seem to be differences between F4 and F7 families. FreeRTOS+TCP includes a modified (hacked?) version of the HAL ETH files for both F4 and F7.

Since I only have access to F7 boards my focus was to get that one working first and asking for help to complete the F4 part. To get started I'd look into the STM32 HAL ETH files in original FreeRTOS+TCP and check for differences.

kikass13, 2021-03-18:
@mikewolfram 
I am not familiar with cube and where their HAL examples are. Do you mean these (probably not?!): 
 * https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_eth.c
 * https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_eth.c

because as far as i can tell, these two are identical. Except for one simple function `static void ETH_Delay(uint32_t mdelay);` which is used for the f4

mikewolfram, 2021-03-18:
It isn't in Cube nor their examples, it's in the FreeRTOS git:
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/main/portable/NetworkInterface/STM32Fxx

kikass13, 2021-03-18:
Sorry i must misunderstood your reply then... 

I already looked a these in the past, but
i cant seem to understand where these files could differ except for the chip specific `_hal_eth.c` files

mikewolfram, 2021-03-18:
You are right, there's only one source file but a few headers per chip.

As far as I remember they use quite a few macros, maybe they do something different?

mikewolfram, 2021-03-18:
Another question, do you get the link-up event? IRQ aren't enabled before.

kikass13, 2021-03-18:
> As far as I remember they use quite a few macros, maybe they do something different?

well i did a diff on the HAL_c files and the HAL_h files and they are essentially the same. And there are no further ifdefs in the freertos+tcp.c file you mentioned so i guess there's nothing more board specific?

kikass13, 2021-03-18:
> Another question, do you get the link-up event? IRQ aren't enabled before.

as far as I remember (i did these test with gdb about two weeks ago) this code (https://github.com/modm-io/modm/blob/develop/ext/aws/modm_lan8720a.cpp#L374-L385) was called properly and the `autoNegotiationFailed` flag was false, which I figured means that configuration of the PHY was possible. 

I can also confirm that 
 * the stack / driver runs normal and other freertos Tasks can run freely without problems.
 * the phy does something because i get a IFUP on my ethernet connection (using the other linux laptop) and the two leds blink whenever something happens on the ethernet rx/tx line
 * What happens is that `ARP` requests (and therefore ICMP packets) are not handled, which means that i do not receive the MAC address of the nucleo.

mikewolfram, 2021-03-18:
In the example is a hook `vApplicationIPNetworkEventHook`, which get's called, when the link is up and everything set up, so it should print the IP etc.

kikass13, 2021-03-18:
Yes that part works.

kikass13, 2021-03-18:
Yes that part works:
 
 * see my example output from the serial port

```
============================
Hello world ... 

Ethernet manager initializing ... 
Initialzing ethernet interface ...
Initializing scheduler
  -- Led blink task run()
  -- Network initialization task run()
  -- Http server task run()
   > IP address: 192.168.1.1
   > Netmask   : 255.255.255.0
   > Gateway   : 0.0.0.0
   > DNS       : 0.0.0.0
```

rleh, 2021-04-13:
Maybe the updates FreeRTOS and FreeRTOS+TCP modules (#566) magically solve the problem? Could you test that @kikass13?

kikass13, 2021-04-14:
@rleh 
I can do that, give me some days to do other stuff first :D

jnewcomb, 2021-04-15:
I feel your pain..
It could be possible the ethernet DMA for the F4 if unable to operate in the 'modm_fastdata' section of memory.
In modm_lan8720a.cpp - edit this:

```
	modm_aligned(32) modm_fastdata
	static DmaDescriptor_t  DmaRxDescriptorTable[RX_BUFFER_NUMBER];

	/* Ethernet Tx DMA Descriptor */
	modm_aligned(32) modm_fastdata
	static DmaDescriptor_t  DmaTxDescriptorTable[TX_BUFFER_NUMBER];
```

to this
```
	modm_aligned(32)
	static DmaDescriptor_t  DmaRxDescriptorTable[RX_BUFFER_NUMBER];

	/* Ethernet Tx DMA Descriptor */
	modm_aligned(32)
	static DmaDescriptor_t  DmaTxDescriptorTable[TX_BUFFER_NUMBER];
```

Let me know if that fixes your issue..

rleh, 2021-04-16:
The Ethernet MAC on STM32F4(29) is indeed not capable of accessing the DTC/CCM memory where the `modm_fastdata` should is located, while this access is possible on STM32F7.

[image]

[image]

---

I just removed the `modm_fastdata` attribute as suggested, but ethernet still does not work.
The data in `DmaTxDescriptorTable` is not actually the DMAed, but the data the pointers in `DmaTxDescriptorTable` point to is also somehow allocated (by FreeRTOS?) in the CCM memory. I disabled the whole CCM memory by modifying the linkerscript and now Ethernet works on my Nucleo-F429 board!

Thank you very much @jnewcomb!

salkinium, 2021-04-16:
`modm_fastdata` is not guaranteed to be DMA-able, but plain `malloc` and `new` are always allocated in DMA-able memory, and this is also true for any malloc that FreeRTOS does. The only other thing in CCM is the main stack, but it's generally bad practice to DMA memory from the stack, so it's probably not that?

(It's probably also not very fast to access the F7 DTCM via the Ethernet MAC, I would place this in normal SRAM too.)

You can use overwrite the option `:platform:heap:allocator=newlib` so that the CCM heap isn't accessible by malloc at all. That would at least rule out the heap.

kikass13, 2021-04-16:
I am not really sure that i understand the solution (my knowledge is way to limited in this regard).
 * Is disabling the CCM a good idea?
 * > I just removed the modm_fastdata attribute as suggested, but ethernet still does not work.
      *  Why does this not work then? 
              * Is the dma now trying to now r/w in sram instead of the CCM ? 
              * And is the data buffer still located in CCM because FreeRTOS created it? (which is CCM for stack/heap by default?)?
 * how can modm ensure that dma is working correctly on different devices then?
     * is there  capability flag in modm, which lbuild can use to define whether or not something has access to specific types memory types?
 *

salkinium, 2021-04-16:
> Is disabling the CCM a good idea?

No, it's 64kB of single-cycle access SRAM (but only from the CPU) it would be very wasteful to ignore that.

> Why does this not work then?

I don't understand it either. Only the main stack, the fastdata and the heap0 is located in CCM. Is there something getting DMA'd from the main stack?

> how can modm ensure that dma is working correctly on different devices then?

The linkerscripts are all specifically designed so that the "normal" data sections (.bss, .noinit, .data) are in main SRAM which is always DMA-able. Same goes for the heap.
Only on devices that have a "fast" memory, like CCM on STM32F4 or TDCM on STM32F7, the data section `.fastdata` is placed there, otherwise it is also placed in main SRAM.

So there should be no surprises like this at all if you don't use any memory attributes, but
if you use *any* memory attribute like `.fastdata` it can lead to weird results, depending on the family and even on the DMA instance.

We need to document this better, since we're starting to look into DMA now more and this very annoying to debug.

> is there capability flag in modm

For the heap there are [these MemoryTraits](https://docs.modm.io/develop/api/stm32f439ziy6/group__modm__architecture__memory.html#ga4ab452f5d1300a45f045af588fc25fb2) that you can use to get memory with extra traits. But you need to specifically request it, by default only DMA-able memory is allocated.

This is also how can allocate memory on the external heap (16MB on the STM32F469 Discovery for example).

```cpp
auto *p = new (modm::MemoryFastData) Foo;
auto *q = new (modm::MemoryExternal, std::nothrow) Foo;
```

kikass13, 2021-04-16:
Thanks for the clarification, one question (which you answered partially) 
 
> So there should be no surprises like this at all if you don't use any memory attributes, but
> if you use _any_ memory attribute like `.fastdata` it can lead to weird results, depending on the family and even on the DMA instance.

My original question was specifically targeted at  dma bus capabilities - how can one (is it plausible?) ensure (via abstraction) that a internal controller/module master has dma access to other device slaves?

Is it (for example) feasible to abstract the Bus matrix shown by @rleh and generate DMA capabilities on a device specific basis: 
 * Device A (stm32f7xxxxx)
    * Master1(memory mapped devic for ETH PHY) > Slave1 (memory mapped device for CCM) > COOL    
 * Device B (stm32f4xxxxx)
    * Master1(memory mapped devic for ETH PHY) > Slave1 (memory mapped device for CCM) > NOT COOL

so that modm (pre compilation?) warn developers that the dma capabilities they have declared can not be  properly mapped and are therefore not possible on the device they are using

rleh, 2021-04-16:
I did some further research: Without changes to the current source code I have to force both the `.fastdata` and the `.stack` sections into RAM (currently in CCM) to get Ethernet running.

Removing the `modm_fastdata` attributed and keeping `.fastdata` section in CCM does not work. I don't understand why, because the `modm_fastdata` or `modm_section(.fastdata)` attribute are not used anywhere else (except `platform/clock/rcc.cpp`).

---

> @salkinium: [...] but it's generally bad practice to DMA memory from the stack [...]

Why?

With Ethernet (and FreeRTOS+TCP) it is probably (?) not possible to allocate all DMA-related variables/buffers statically, so we have to choose between stack and heap.
In some applications the usage of heap is not desirable (and with FreeRTOS config `configSUPPORT_DYNAMIC_ALLOCATION`), so stack is the only option...

rleh, 2021-04-16:
> My original question was specifically targeted at dma bus capabilities - how can one (is it plausible?) ensure (via abstraction) that an internal controller/module master has dma access to other internal device slaves?

For dynamically allocated memory: Via memory traits as explained by @salkinium above.
Stack and static memory: I'm not aware of any solution to the problem other than not using storage like CCM at all...

We could provide a utility function in modm that displays the capability of a given memory address (pointer) at runtime, but that would only be useful in rare situations (like now 😆).

salkinium, 2021-04-16:
> We could provide a utility function in modm 

I mean, we could have a debug assertion in the DMA driver to check the address range at runtime. The information is currently **not** part of modm-devices, but it's not difficult to add. It's probably the best quick hack that we can add.

chris-durand, 2021-04-16:
> > @salkinium: [...] but it's generally bad practice to DMA memory from the stack [...]
> 
> Why?

If you return from the function the stack memory gets deallocate. You have to make sure the DMA transfer could never continue afterwards.

rleh, 2021-04-16:
> If you return from the function the stack memory gets deallocated. You have to make sure the DMA transfer could never continue afterwards.

Of course. But thats the same issue with (manually) freeing dynamically allocated memory...

rleh, 2021-04-16:
I think I would consider it useful to give the modm:platform:cortex-m module an option not to create the stack in non-DMA memory, and probably even enable it by default.
(As far as I know the stack is also in SRAM by default on other toolchains (ST-Cube...)).
If you want to have the additional performance of TCM/CCM for your stack you can enable this option and should be aware of the disadvantages.

A broader use of DMA probably brings more advantages in total than having the stack in TCM/CCM.
When I look at DmaSpiMaster (which has recently become available on many more devices) it is (unfortunately) very easy to run into the same problem there unknowingly.

salkinium, 2021-04-16:
>> @salkinium: [...] but it's generally bad practice to DMA memory from the stack [...]
>
> Why?

I was mainly thinking about modm's use of protothreads and resumable functions, where the stack doesn't really exists and is continously changed under the hood.

For FreeRTOS it's different, since the stack remains untouched IFF you wait synchronously for the DMA transfer to end. In general though, I'm not sure what guarantees C/C++ gives about stack memory lifetimes.

Regardless, only the *main stack* (which is not a FreeRTOS thread!) is allocated in the CCM, all other stacks are either dynamically allocated in heap, or statically (by modm::Thread) in main SRAM DMA-able memory. 

I am however open to changing the main stack placement, but for CCM the access speedup is >2x compared to SRAM, therefore I placed it there by default.

> Without changes to the current source code I have to force both the .fastdata and the .stack sections into RAM (currently in CCM) to get Ethernet running.

You really have to go out of your way to allocate something in fastdata, which is why this is a very confusing to me. `scons listing` will give you a map of what objects are placed in `.fastdata` etc.

rleh, 2021-04-16:
> I mean, we could have a debug assertion in the DMA driver to check the address range at runtime.

👍🏽 

> The information is currently **not** part of modm-devices, but it's not difficult to add. It's probably the best quick hack that we can add.

Where does the information for the `MemoryTraits modm::MemoryDMA` come from?

salkinium, 2021-04-16:
> Where does the information for the MemoryTraits modm::MemoryDMA come from?

It's shamefully hardcoded inside the linkerscripts in the heap table (in manually translated *binary* format even!!!). This needs to be extracted into modm-devices ideally.
https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/stm32/linkerscript/stm32_dccm.ld.in#L59-L68

chris-durand, 2021-04-16:
>  In general though, I'm not sure what guarantees C/C++ gives about stack memory lifetimes.

It's simple, in C++ terms objects of *automatic storage duration* (that is all local objects *not* declared as `static`, `extern` or `thread_local`) live exactly as long as the enclosing code block.

mikewolfram, 2021-04-19:
Reading your posts about the .fast_data section reminds me, there was a comment in the original STM32 HAL implementation:
```c++
/* MAC buffers: ---------------------------------------------------------*/

/* Put the DMA descriptors in '.first_data'.
This is important for STM32F7, which has an L1 data cache.
The first 64KB of the SRAM is not cached.
See README.TXT in this folder. */

/* Ethernet Rx MA Descriptor */
__attribute__ ((aligned (32)))
#if defined(STM32F7xx)
        __attribute__ ((section(".first_data")))
#endif
        ETH_DMADescTypeDef  DMARxDscrTab[ ETH_RXBUFNB ];

#if( ipconfigZERO_COPY_RX_DRIVER == 0 )
        /* Ethernet Receive Buffer */
        __ALIGN_BEGIN uint8_t Rx_Buff[ ETH_RXBUFNB ][ ETH_RX_BUF_SIZE ] __ALIGN_END;
#endif

/* Ethernet Tx DMA Descriptor */
__attribute__ ((aligned (32)))
#if defined(STM32F7xx)
        __attribute__ ((section(".first_data")))
#endif
        ETH_DMADescTypeDef  DMATxDscrTab[ ETH_TXBUFNB ];

#if( ipconfigZERO_COPY_TX_DRIVER == 0 )
        /* Ethernet Transmit Buffer */
        __ALIGN_BEGIN uint8_t Tx_Buff[ ETH_TXBUFNB ][ ETH_TX_BUF_SIZE ] __ALIGN_END;
#endif
```
Good finding and sorry, I didn't add it to the first implementation. Since I'm on a F7 and had no access to an F4 for testing I skipped all non-F7 specific code at that time.