modm-io/modm #617: Outdated CI Docs

TomSaw, 2021-04-20:
There's old CircleCI Docu in test/README.md. Please update instructions for new github-CI

```
!!! tip "Add your example to our CI"
	If you created a new folder inside `modm/examples`, you need to add it to
	one of the jobs in `.circleci/config.yml`:
	```sh
	- run:
	  name: Examples YOUR TARGET
	  command: |
	    (cd examples && ../tools/scripts/examples_compile.py new_folder)
	```

```

rleh, 2021-04-20:
```diff
+one of the jobs in `.github/workflows/linux.yml`:
-one of the jobs in `.circleci/config.yml`:
````

rleh, 2021-04-20:
@TomSaw DO you want to fix that yourself (one additional commit in one of you open pull requests)?

TomSaw, 2021-04-20:
Oh... thought it's more complicated.
Never touched any CI before. If it's just 3 lines, would leave it to someone who already knows.. got enough construction sites for now.

rleh, 2021-04-20:
Is `test/README.md` visible on https://modm.io/? (If yes: where?) @salkinium 

Otherwise we (someone™️) should rewrite the MkDocs syntax  (e.g. `!!! tip`) to something that looks nice in Githubs Markdown (currently it looks a bit ugly: https://github.com/modm-io/modm/blob/develop/test/README.md)

salkinium, 2021-04-20:
Whoops. I forgot about rewriting this when I adapted the homepage guides. I'll rewrite it when I have time.

PR #619 (2021-04-21): Fix RTT discover and test docs
Apparently the RTT identifier string was placed somewhere on the stack during copy, thus OpenOCD found the wrong RTT location. This fixes it by using `constinit` for force the initialization into the `.data` section only, without calling a constructor! ~~writing the `.rtt` part as a word from a literal pool in the constructor.~~

This also fixes #617. cc @TomSaw @rleh
closed 2021-04-21 by salkinium