modm-io/modm #625: Bootloader support

jnewcomb, 2021-04-28:
Hello folks,
I'm trying to implement a FLASH resident bootloader. What are people's thoughts on the best way? 
Obviously not my way.. but for completeness, I'll describe my nugatory efforts so far...

I've generated 2 separate modm builds - first call 'bootloader', second called 'app'.
bootloader resides in the first 128K and the 'app' linker file shifts everything up by the same amount.
Both projects work as expected  - the bootloader jumps to the application codes reset handler... But, in order to do this, I have to download it separately - bootloader first then application code. Ideally, I want to pull the precompiled bootloader (as a static library?) into the application - and maybe even get it working on the debugger. Need to be combined in the hex file.
 To this end I've prepared the app linker file:
```
MEMORY
{
	BOOT (rx) : ORIGIN =    0x08000000, LENGTH = 128K
	FLASH (rx) : ORIGIN =   0x08020000, LENGTH = 1920K
...
}
...
ENTRY(Reset_Handler)
...
SECTIONS
{
	.vector_bootloader :
	{
		. = ALIGN(4);
		__vector_table_bootloader_start = .;
		KEEP(*(.vector_bootloader))
		/* https://stackoverflow.com/questions/42295298/place-segments-of-external-static-library-to-specific-locations */
		*libbootloader_lib.a:*(.vector_bootloader)
		. = __bootrom_size__;
		. = 0x20000;
		__vector_table_bootsection_end = .;
	} >BOOT

	.vector_rom :
	{
		. = ALIGN(4);
		__vector_table_rom_start = .;
		__vector_table_ram_load = .;
		/* Initial stack address, Reset and NMI handler */
		KEEP(*(.vector_rom))
	} >FLASH
...
}
```

I have added the following to the bootloader MakeFileLists:
```
## Added by JN create a static library (*.a) for locating in the application code bootloader sector region
add_library(${CMAKE_PROJECT_NAME}_lib STATIC ${APP_SOURCE_FILES})
target_include_directories(${CMAKE_PROJECT_NAME}_lib PUBLIC include)
target_link_libraries(${CMAKE_PROJECT_NAME}_lib
  modm_options
  modm_warnings
  modm)
```
This generates a bootloader_lib.a
My aim is to pull this into the application. I have added the following to the application MakeFileLists:
```
target_link_libraries(${CMAKE_PROJECT_NAME}
  modm_options
  modm_warnings
  modm
  "C:/Projects/modm/examples/project/bootloader/build/VisualGDB/Debug/libbootloader_lib.a"
  )
```
(there are better ways - find_library() etc, but I just want to see it working before the tidy up)

From the *.map file below, obviously, it's not worked.. I've come from IAR hence my appaling use of CMake!
Are there any hints on this to be shared?
Should I make a 3rd CMakeLists and include the other two?
Or - Can anyone share the relevant CMake/linker cut & pastes for a simple FLASH resident bootloader?
[... 124 lines cut ...]

.heap0          0x10000c08     0xf3f8 load address 0x080201b4
 *(.heap0)
                0x10000c08                . = ALIGN (0x4)
                0x10000c08                __heap0_start = .
                0x10010000                . = (ORIGIN (CCM) + LENGTH (CCM))
 *fill*         0x10000c08     0xf3f8 
                0x10010000                __heap0_end = .

.text           0x080201c0     0x2b7c
 *(.text .text.* .gnu.linkonce.t.*)
 .text._ZN4modm8IODevice5writeEPKc
                0x080201c0       0x3e CMakeFiles/application.dir/board.cpp.obj
                0x080201c0                modm::IODevice::write(char const*)
 .text._ZN4modm8IOStreamC2ERNS_8IODeviceE
                0x080201fe       0x24 CMakeFiles/application.dir/board.cpp.obj
                0x080201fe                modm::IOStream::IOStream(modm::IODevice&)
                0x080201fe                modm::IOStream::IOStream(modm::IODevice&)
 .text._ZN4modm8IOStreamlsERKm

                0x08020222       0x20 CMakeFiles/application.dir/board.cpp.obj
                0x08020222                modm::IOStream::operator<<(unsigned long const&)
 .text._ZN4modm8IOStreamlsEPKc
                0x08020242       0x28 CMakeFiles/application.dir/board.cpp.obj
```

salkinium, 2021-04-28:
I'm not sure you want to have one big hex/binary file, since the bootloader is probably not 128k large, so if you want to have one continuous binary, there'll be a lot of zeros in there until the application starts.

You could however call OpenOCD manually and program both ELFs in one go:

```
openocd -f modm/openocd.cfg -c "program path/to/bootloader.elf verify" -c "program path/to/application.elf verify reset exit"
# or the binaries
openocd -f modm/openocd.cfg -c "program path/to/bootloader.bin verify 0x08000000" -c "program path/to/application.bin verify reset exit 0x08020000"
```

I'm not sure about debugging both apps at the same time, it would be **really** neat if you made that work though.
Intuitively you would have to merge the two ELF files incl. debug information perhaps with a Python script using `pyelftools`.
You may also have to rewrite relative file paths to distinuguish `bootloader/main.cpp` from `application/main.cpp`.
Perhaps you could then also just pass that shared ELF file to openocd, and it'll automatically program the *two* applications for you?

salkinium, 2021-04-28:
Just FYI: there's an [option for flash offsets](https://modm.io/reference/module/modm-platform-cortex-m/#linkerscriptflash_offset) so you don't have to create your own application linkerfile.

jasa, 2021-04-29:
The CMake scripts is not relying on start-group and end-group linker flags, so creating static libraries containing modm itself is dangerous. Unreferenced objects might disappear since "used" attribute is not specified in the source and linker tables is also a problem. (eg. **hardware init** and **assert**)

jnewcomb, 2021-04-30:
Thanks for all your advice guys..
In the end.. I created a new sector in the application link script called vector_bootloader and merged boot binary into it.
````
  add_custom_command(TARGET ${project_name}
    POST_BUILD
	# JN: Although I've done my best to FILL() gaps in the linker script, they still creep in.
	# Seen this in sector .heap0 (NOLOAD), where FLASH seems to be allocated, but not initialised with a value i.e unprogrammed. This
	# can be seen as 'gaps' in the HEX files. Not good for calculating a reliable checksum!!!.. So.. the workaround is this..
	# Force fill 'unallocated' from 0x00 to 0xFF to match 'unprogrammed' FLASH. It's all rather insane if you ask me!
	# All this makes calculating and comparing checksums near impossible. We try to patch this up with OBJCOPY..
	# (This will overwrite the ELF file with a fixed one)
	COMMAND ${CMAKE_OBJCOPY} --gap-fill=0xff  ${project_name}.elf 

	# Here we need to combine the bootloader with the application. vector_bootloader will just get overwritten
	COMMAND ${CMAKE_OBJCOPY} --update-section .vector_bootloader="C:/Projects/modm/examples/project/bootloader/build/VisualGDB/Debug/bootloader.bin" ${project_name}.elf
	
	# Create a complete BIN file
	COMMAND ${CMAKE_OBJCOPY} --output-target=binary ${project_name}.elf ${project_name}.bin

	# Convert ELF file for external programming
    COMMAND ${CMAKE_OBJCOPY} --output-target=srec --srec-len 0x20 --srec-forceS3  ${project_name}.elf ${project_name}.srec
    COMMAND ${CMAKE_OBJCOPY} --input-target=srec --output-target=ihex ${project_name}.srec ${project_name}.hex
    )	
````

salkinium, 2021-04-30:
I'm closing this as this solution is not something we can integrate into modm.

closed 2021-04-30 by salkinium