modm-io/modm #628: CMake on Windows: Unescaped backslashes

jensboe, 2021-05-07:
Hi,
I am using Windows and CMake and just start using modm. If I like modm I will start a PR to fix this issue by myself but for now I just reporting.

The generated repo.cmake contains unescaped backslashes.
`    ${CMAKE_CURRENT_LIST_DIR}/src\modm\architecture\driver\atomic\flag.cpp`

CMake on Windows accept slashes or escaped backslashes. I don't know the common preferences but i preferer normal slashes.

jensboe, 2021-05-07:
The set of  CCFLAGS makes also trouble.
After removing the, after removing file-prefix-map a can create hex files.
```
-ffile-prefix-map=${MODM_GCC_PATH}=. \
# -ffile-prefix-map=${CMAKE_CURRENT_SOURCE_DIR}=. \
```

jensboe, 2021-05-07:
The set of  CCFLAGS makes also trouble, after removen ```
-ffile-prefix-map=${MODM_GCC_PATH}=. \
# -ffile-prefix-map=${CMAKE_CURRENT_SOURCE_DIR}=. \
```

salkinium, 2021-05-07:
Hm, Windows support is difficult to get right…

There is a new CMake generator, see #568. Perhaps it solves some of these issues.

jensboe, 2021-05-08:
It only generates other issues.
I hate CMake. I only use itbecause the working interaction between VSCode Intellisense and the compilerswitches.
I think it's easier to rewrite some code(year, python) for sync between VSCode and SCons(which works) or switching to WSL2.

rleh, 2021-05-08:
> VSCode Intellisense

For me VSCode works great (with the C++/Intellisense plugin) using SCons (or even without a build system) if I open the project folder (with the `project.xml` file) in VSCode. I then use the build-in terminal for compiling, programming and debugging...

salkinium, 2021-05-08:
I'm sorry for your troubles, I also dislike CMake a lot…

Perhaps you can try to import this project, which I've just tested to work on VSCode with CMake (but on macOS):
https://github.com/modm-ext/modm_starter_project

rleh, 2021-05-10:
Maybe we should use [`FindPython3`](https://cmake.org/cmake/help/latest/module/FindPython3.html) instead of assuming a python3 executable is in $PATH?

salkinium, 2021-05-10:
> I think the call of size.py isn't correct.

There's an extra `src/` in the path, because it's designed for the sources to be in a `src/` folder unlike our examples. This should be switched based on whether the `include_cmakelists` option is set to true.

jensboe, 2021-05-10:
> > After that a "open file with" dialog opens up and and after selecting a program, the cmd says failed.
> > I think the call of size.py isn't correct.
> 
> Probably both because "CMake wasn't able to find python3".
> (I assume you have python3 installed on you windows system, otherwise I wonder how lbuild works...)

I assume that too. Python 3 is installed automatically after the first call with Windows store. (I don't know who and why somebody at MS think that that is a got idea)

jensboe, 2021-05-10:
Replacing all occurrences of `${PYTHON3_EXECUTABLE}` with `python` leads to a clean exitcode 0 build. :)

I think CMake itself is the problem now.

salkinium, 2021-05-10:
Could Windows perhaps not install `python3`, only `python`?

```
find_package(Python COMPONENTS Interpreter REQUIRED)
```

jensboe, 2021-05-10:
> I don't know who and why somebody at MS think that that is a good idea

I's a bad idea. After install Python without the windows store app function and cleaning the path:

```
[cmake] -- Found Python3: C:/Python/Python39/python.exe (found version "3.9.5") found components: Interpreter 
```

On a fresh python installation pyelftools is needed but not mention (only for old ubuntu versions)

jensboe, 2021-05-10:
> Could Windows perhaps not install `python3`, only `python`?
> 
> ```
> find_package(Python COMPONENTS Interpreter REQUIRED)
> ```

That didn't set `PYTHON3_EXECUTABLE`.

`find_program(PYTHON3_EXECUTABLE python REQUIRED)` works now, too.

jensboe, 2021-05-10:
> Could Windows perhaps not install `python3`, only `python`?

The 'windows store python' did both. The 'real python' only python.

To sum this evening:
1) Backslashes in `.\modm\cmake\ModmConfiguration.cmake` must be normal slashes (or must be escaped?). I think slashes are the nicer way because their compatibility to unix systems.
2) `pyelftools` must be mention in the getting started guide for "windows and cmake only" (BTW: Why are they installed in newer ubuntu or are they just a dependency of Scons?)
3) A nice way for python3 (non-windows) and python (windows) command must be found.

sbx320, 2021-05-12:
I've looked a bit deeper into the Windows Store Python issues. 

The Windows Store Python does something very odd. Rather than adding the python install folder to the PATH, `%localappdata%/Microsoft/WindowsApps` is added to the PATH, which contains zero byte "executables" (such as `python.exe`). Launching those executables causes the corresponding store package to be used, very similar to a symlink (yet it is not a proper symlink). This works just find for launching programs (which is also why replacing `PYTHON3_EXECUTABLE` with `python`works as a workaround). However CMake does a bit more:

CMake's `find_program` walks all $PATH entries and checks whether the requested file exists in that directory. To validate whether a file exists, it first reads the file attributes and then - if the attributes indicates a NTFS reparse point [1] - it attempts to create a handle to that file to validate that the file at the end of the reparse point exists [0]. However this fails because you cannot create regular handles to those magic zero byte redirection files. Instead you can only execute them. 

So I suppose that part is an upstream bug of either CMake or Windows rather than a modm bug. 

--- 

[0]: See https://github.com/Kitware/CMake/blob/master/Source/kwsys/SystemTools.cxx#L1336
[1]: NTFS Reparse Points are essentially entry points for filesystem filters, which are then used to implement things like symlinks or mount point. Microsoft enjoys using those to bolt on various new things to NTFS.

jasa, 2021-11-19:
I have fund the following issues:
* Python installed from python.org works. The "Windows Store" version of Python does not work with CMake. 
* CMake requires that Python has the same build architecture, don't mix 32 and 64 bits versions
* Backslashes: lbuild uses os.path and maybe a major rewrite to pathlib is needed. (not examined properly) but workaround exists, see below.

Tested with following version:
[CMake 3.22](https://github.com/Kitware/CMake/releases/download/v3.22.0/cmake-3.22.0-windows-x86_64.msi)
[Python 3.9.9](https://www.python.org/ftp/python/3.9.9/python-3.9.9-amd64.exe)
[Ninja 1.10,2](https://ninja-build.org/) (instead of nmake)
[arm-none-eabi 10.3](https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-win32.exe)

This ain't pretty but it works. The following code can be placed in the main CMakeLists.txt and it will replace the backslashes automatically. 
```cmake
function (convert_backslash_path FILE_NAME)
  file(READ ${FILE_NAME} SRC)
  file(TO_CMAKE_PATH ${SRC} DST)
  file(WRITE ${FILE_NAME} ${DST})
endfunction()

convert_backslash_path("modm/CMakeLists.txt")
convert_backslash_path("modm/cmake/ModmConfiguration.cmake")
```

CrustyAuklet, 2022-02-05:
I have been playing with modm on windows recently and also ran into this issue.

I use CMake a lot across different platforms and IMO one should always use `/` path seperators, even on windows! In fact I use Windows daily at work and I don't think I have ever really used a backslash. Powershell, CMake, etc all work fine with `/` and it keeps me sane to have things the same across systems.

As far as the weird windows store python, one of the first things I do when I have a fresh windows install is remove all those executables and remove that path from my PATH. Then I install the "real" python from python.org. Again, it's a sanity thing.

The changes I made to address things, taking into consideration the above two things, you can see in my modm fork here: https://github.com/CrustyAuklet/modm/commit/bf542fb263f9f5dbbe4ccb29249591ae572b2588

I kept changes to the python restricted to cmake generation code, and in the CMake I check for `python` in addition to `python3` if it's windows. I could figure out some more complex checking logic if you want to support more edge cases? I have used the [FindPython3](https://cmake.org/cmake/help/latest/module/FindPython3.html) module before and it is probably a better choice.

salkinium, 2022-02-08:
Feel free to change the CMake generator and path issues as you see fit! My only experience with Windows is inside a VM and only briefly, so these issues are really not intentional.

(Unfortunately changing lbuild to use pathlib exclusively breaks backwards compatibility since `pathlib.Path` doesn't behave like a string everywhere. It's been bugging me, but I can't change it anymore…)

jensboe, 2023-11-05:
This issue can be closed. pathes are generated with `/` now.
I have some other issues with Cmake now but I think its better to create a new issue for that.

closed 2023-11-05 by jensboe