modm-io/modm #630: Stream formatting - hex values prefixed

jnewcomb, 2021-05-13:
Hi folks..
Try as I might - I'm unable to get the logger to print a 32bit HEX..
I've run out of ideas on the code.. any hints as to where to look next? (another career maybe!)

Based on the STM32F429 blink example, GCC 10.2.1

````
auto crc_val1 = UINT32_C(32768);
MODM_LOG_DEBUG << "# crc_val1: " << crc_val1 << modm::endl;
MODM_LOG_DEBUG << "# crc_val1: " << std::hex << crc_val1 << modm::endl;

auto crc_val2 = INT32_C(32769);
MODM_LOG_DEBUG << "# crc_val2: " << crc_val2 << modm::endl;
MODM_LOG_DEBUG << "# crc_val2: " << std::hex << crc_val2 << modm::endl;

uint32_t crc_val3 = UINT32_C(32770);
MODM_LOG_DEBUG << "# crc_val3: "  << crc_val3 << modm::endl;
MODM_LOG_DEBUG << "# crc_val3: " << std::hex << crc_val3 << modm::endl;

uint32_t crc_val4 = UINT32_C(32771);
MODM_LOG_DEBUG << "# crc_val4: "  << crc_val4 << modm::endl;
MODM_LOG_DEBUG << "# crc_val4: " << std::hex << crc_val4 << modm::endl;

uint32_t crc_val5 = UINT32_C(32772);
MODM_LOG_DEBUG << "# crc_val5: "  << static_cast<uint32_t>(crc_val5) << modm::endl;
MODM_LOG_DEBUG << "# crc_val5: " << std::hex << static_cast<uint32_t>(crc_val5) << modm::endl;

uint64_t crc_val6 = UINT32_C(32773);
MODM_LOG_DEBUG << "# crc_val6: "  << crc_val6 << modm::endl;
MODM_LOG_DEBUG << "# crc_val6: " << std::hex << crc_val6 << modm::endl;

MODM_LOG_DEBUG << "# CRC OK: " << UINT32_C(0xABCDEF) << modm::endl;
MODM_LOG_DEBUG << "# CRC OK: " << std::hex << UINT32_C(0xABCDEF) << modm::endl;

MODM_LOG_DEBUG << "# CRC OK: " << static_cast<uint32_t>(UINT32_C(0xABCDEF)) << modm::endl;
MODM_LOG_DEBUG << "# CRC OK: " << std::hex << static_cast<uint32_t>(UINT32_C(0xABCDEF)) << modm::endl;

MODM_LOG_DEBUG << "# CRC OK: " << std::hex << 123 << modm::endl;
MODM_LOG_DEBUG << "# CRC OK: " <<  123 << modm::endl;

MODM_LOG_DEBUG << "# CRC OK: " << 1.23 << modm::endl;
````
gives:

````
# crc_val1: 32768
# crc_val1: 0x10000B2832768
# crc_val2: 32769
# crc_val2: 0x10000B2832769
# crc_val3: 32770
# crc_val3: 0x10000B2832770
# crc_val4: 32771
# crc_val4: 0x10000B2832771
# crc_val5: 32772
# crc_val5: 0x10000B2832772
# crc_val6: 32773
# crc_val6: 0x10000B2832773
# CRC OK: 11259375
# CRC OK: 0x10000B2811259375
# CRC OK: 11259375
# CRC OK: 0x10000B2811259375
# CRC OK: 0x10000B28123
# CRC OK: 123
# CRC OK: 1.23000e+00
````

Takes me here iostream.hpp (line 191) - It seems to think it's a function pointer..
````
	/// Write the hex value of any function pointer
	template<class Ret, class... Args>
	IOStream&
	operator << (Ret(*pointer)(Args...))
	{ writePointer(reinterpret_cast<const void*>(&pointer)); return *this; }
````

salkinium, 2021-05-13:
You need to use `modm::hex`, I'm not even sure what std::hex actually does… 😬

salkinium, 2021-05-14:
> Error format '%X' expects argument of type 'unsigned int', but argument 3 has type 'long unsigned int'

`unsigned int` and `long unsigned int` are separate types, that are both 32-bit unsigned integers (on Cortex-M). The printf specification wants an `long unsigned int` for `%X`, however `uint32_t` is aliased to `unsigned int`. 😒

There's no way to properly fix this, you may be able to ignore the printf compile error by adding this to the project.xml compile flag collectors:

```xml
<collectors>
  <collect name="modm:build:ccflags">-Wno-error=format</collect>
</collectors>
```

> any hints as to where to look next? (another career maybe!)

I feel you!

> Why is it the more I learn, the dumber I feel!

It's not you, it's the standards body that has for many decades ignored how embedded systems work in practice and now you cannot change it anymore. Short of changing to Rust there's nothing that can be done. At least in recent years embedded has taken a more dominant role in the C++ standards body, but it's a little late now…

jnewcomb, 2021-05-14:
Well.. `modm::hex` and ` "%lX"` fixed that one!  The joys of starting a new codebase.
Thanks for clearing up my confusion regarding the use of int types, all makes sense now!

closed 2021-05-14 by jnewcomb