modm-io/modm #633: ITM does not work anymore with OpenOCD HEAD / DEPRECATED 'tpiu config'

strongly-typed, 2021-05-19:
Hi there,

I have [working config](https://github.com/modm-io/modm/issues/501) for logging with ITM. Works great with current modm and OpenOCD `0.11.0` from homebrew.

Something in OpenOCD is about to change:

```
➜  src git:(master) ✗ scons log-itm fcpu=80000000
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
╭───OpenOCD───> Single Wire Viewer
╰─────SWO────── stm32l432kbu6
Open On-Chip Debugger 0.11.0+dev-00158-gd3a859cc4 (2021-05-19-18:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : DEPRECATED target event trace-config
modm_program
Info : DEPRECATED 'tpiu config' command: Adding a TPIU 'stm32l4x.cpu.tpiu' in the configuration
Info : DEPRECATED 'tpiu config' command: Running: 'tpiu create stm32l4x.cpu.tpiu -dap stm32l4x.dap -ap-num 0'
invalid command name "stm32l4x.cpu.tpiu"
Info : DEPRECATED 'tpiu config' command: Running: 'stm32l4x.cpu.tpiu configure -protocol uart -traceclk 80000000 -output /var/folders/20/hy2sbkcn34l3c7gv65shd7ww0000gn/T/tmpo840owbt -formatter off'
Info : DEPRECATED 'tpiu config' command: Running: 'stm32l4x.cpu.tpiu enable'
Error executing event trace-config on target stm32l4x.cpu:
/usr/local/Cellar/open-ocd/HEAD-d3a859c/bin/../share/openocd/scripts/mem_helper.tcl:6: Error: invalid command name "mem2array"
in procedure 'modm_itm_log'
in procedure 'mmw' called at file "/usr/local/Cellar/open-ocd/HEAD-d3a859c/bin/../share/openocd/scripts/target/stm32l4x.cfg", line 119
in procedure 'mrw' called at file "/usr/local/Cellar/open-ocd/HEAD-d3a859c/bin/../share/openocd/scripts/mem_helper.tcl", line 36
at file "/usr/local/Cellar/open-ocd/HEAD-d3a859c/bin/../share/openocd/scripts/mem_helper.tcl", line 6
^Cscons: *** [log_openocd_itm] Build interrupted.
scons: Build interrupted.
scons: building terminated because of errors.
scons: writing .sconsign file.
```

Is anyone aware of the upcoming changes? What needs to be changed in `modm` or do we need to report a bug to OpenOCD?

salkinium, 2021-05-19:
Oh, indeed the API changed completely: http://openocd.org/doc/html/Architecture-and-Core-Commands.html (near the bottom)
Looks fairly complicated… 

This is what needs to be changed in modm if you want to figure out how to fix it: https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/openocd.cfg.in#L12

salkinium, 2021-05-19:
Nvm, the deprecation message contains the new commands, it seems to fail on the unrelated mem2array command.

salkinium, 2021-05-20:
Do you want to fix this @strongly-typed?

closed 2021-05-26 by salkinium

strongly-typed, 2021-05-27:
Thanks so much! Was able to verify that it works and added simple [blue-pill example here.](https://github.com/modm-io/modm/pull/636)

PR #635 (2021-05-26): [openocd] Upgrade deprecated TPIU+ITM configuration
Fixes #633.

You now need to manually call `Itm::initialize()` since OpenOCD does not do it anymore.

Tested in hardware with the NUCLEO-F103RB ITM example.

cc @strongly-typed