modm-io/modm #641: delay_ns off by about factor 10^2

XDjackieXD, 2021-06-12:
Hi!

I just noticed that delay_ns is off in my projects by about 10² with code that was working fine with the same version of modm and the exact same code. Only the toolchain (GCC, newlib and openocd) were updated in the meantime (delay_ns(360); took roughly the right amount of time previously and now takes around 360us. delay_ns(300000); now takes around 33ms).

I will do further tests to reproduce this and will try to downgrade my toolchain versions and create a minimal project that triggers this bug and report back.

XDjackieXD, 2021-06-12:
https://chaosfield.at/s/testproject.tar.xz here is a test project that reproduces this behaviour with the latest modm version.
My GCC version is 11.1.0, Binutils is 2.36.1 and Newlib is 4.1.0.

rleh, 2021-06-12:
> My GCC version is 11.1.0

Which GCC variant are you using?

The latest `arm-none-eabi-gcc` ("official" by ARM) is GCC 10 from december 2020: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

Since we had problems in the past with GCC variants of various Linux distributions or self-compiled variants, our [installation guide](https://modm.io/guide/installation/) recommends the [GCC toolchain provided by ARM](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads).
Could you please try to reproduce your issue with this toolchain?

XDjackieXD, 2021-06-12:
I was using arm-none-eabi-{gcc,gdb,newlib,binutils} from the arch repo. I uninstalled those and used the latest GCC ARM toolchain you linked with the exact same results. 9.6ms instead of 100us with the testproject I linked above.
The STM is running at the correct speed though as both CAN and UART are working correctly

XDjackieXD, 2021-06-12:
The weird thing is that exactly the modm version and code was working fine a few months earlier. I just came back to it after a fresh rollout of our workstations (and the system update that came with it).
So it can only be some part of the host system that causes this (arm toolkchain, lbuild, openocd, ...)

salkinium, 2021-06-12:
> When I use the delay_us function instead everything works as expected.

delay_us/ms on F3 converts the delay to cycles and then uses the [hardware cycle counter](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/delay.cpp.in#L20) to count those, which is more robust against interrupts. (Though it falls back to software counter for Cortex-M0 and Cortex-M7).

However, the delay_ns is a [pure software counter](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/delay_impl.hpp.in#L34), optimized for lowest overhead possible for short delays but its not very accurate. This is the reasons that using the delay_ns methods with values >1000 is not recommended. But a factor of 100 is a bit much though, I was hoping more like 2-4x offset max.

I have two suspicions:
1. The inline asm syntax may be translated differently by GCC11 than GCC10. You should check the resulting asm with the debugger and step through it.
2. The value of `modm::platform::delay_ns_per_loop`, should at [64MHz on F3 be 3000/64 = 46](https://github.com/modm-io/modm/blob/develop/src/modm/platform/clock/stm32/rcc_impl.hpp.in#L103).

XDjackieXD, 2021-06-12:
Yeah I was using it for a ~300ns delay and my program had drastically dreased performance because it was waiting 30us instead on every occurence which sums up quickly.
I'm now using 1us and it's working nicely.
The high value in the test project was purely for demonstration.

1. this is the disassembly of the delay_ns function with the official ARM GCC 10.2:

```
   0x10000198 <+8>:	ldr	r3, [pc, #28]	; (0x100001b8 <modm::delay_ns(unsigned long)+40>)
   0x1000019a <+10>:	ldrh	r2, [r3, #0]
   0x1000019c <+12>:	ldr	r3, [r7, #4]
   0x1000019e <+14>:	movs	r1, #10
   0x100001a0 <+16>:	muls	r3, r2
   0x100001a2 <+18>:	subs	r3, r3, r1
   0x100001a4 <+20>:	subs	r3, r3, r2
   0x100001a6 <+22>:	bpl.n	0x100001a4 <modm::delay_ns(unsigned long)+20>
   0x100001a8 <+24>:	str	r3, [r7, #12]
=> 0x100001aa <+26>:	nop
   0x100001ac <+28>:	adds	r7, #20
   0x100001ae <+30>:	mov	sp, r7
   0x100001b0 <+32>:	ldr.w	r7, [sp], #4
```

2. I checked the value using gdb and its 47.

But the oscilloscope still says the periodic time of the output square wave is 9.6ms so one of the delays is 4.8ms which is way higher than the 100us it should be.

salkinium, 2021-06-12:
Hm, I see the code has been placed into RAM due to the `modm_fastcode` attribute.
Try removing the `modm_fastcode` attribute from the function and see if that fixes the problem.

I found a Nucleo-F303K8 and I'll try to reproduce this.

salkinium, 2021-06-12:
I can reproduce this on my Nucleo, removing `modm_fastcode` doesn't help though. Investingating…

salkinium, 2021-06-12:
Stepping through the loop:

1.  `muls	r3, r2`: r3 = 100'000, r2 = 47 -> r3 = 4'700'000
2.  `subs	r3, r3, r1`: r1 = 10 -> r3 = 4'700'000
3. `subs	r3, r3, r2`: r2 = 47 -> Loops ~100.000 times… which is just wrong?

Ok, there's a mistake here: I think I meant to `multiply the overhead cycles with the ns per cycle` as the comment says, I'm not sure why this was changed. Changing the muls instruction to  `"muls.n %[ovhd], %[ovhd], %[dnpl] \n\t"` fixes the 100x issue, being accurate for 100us, 10us, 1us, but not <1000ns for me, so the overhead thing is still wrong.

There's also another issue: If the delay_ns gets inlined, it's placed into flash, which takes 6 cycles per loop, if its placed in CCM RAM it's only 3 cycles per loop. `__attribute__((noinline))` forces it into CCM RAM, however, it incurs an additional jump overhead, which makes very low delays impossible, while inlining makes the overhead unpredictable. Hmpf.

PR #642 (2021-06-16): [core] Fix delay_ns cycle calculation
There's [a massive error in the algorithm since we muliply the nanoseconds with `modm::platform::delay_ns_per_loop` first for some reason](https://github.com/modm-io/modm/issues/641#issuecomment-860099236). I'm honestly shocked at just how wrong this is?

- [x] Fix loop count for all STM32
- [x] Fix overhead computation for <1000ns
- [x] Fix inlining of delay_ns function
- [x] Some automatic or semiautomatic testing of cycles via `DWT->CYCCNT` or `SysTick->VAL`
- [x] Tested in hardware
- [x] Enable DWT for Cortex-M7 (missing unlock key)
- [x] Extend `CYCCN [... cut]
closed 2021-06-16 by salkinium