modm-io/modm #647: Advice: Options for internal messaging with a point-to-point UART topology

jnewcomb, 2021-07-03:
Hi folk,
If you have a chance, you shared insights would be greatly appreciated.

My hardware platform has a PC and 3 micros chained together with point-to-point UART connections. (i.e not multidrop)
  Host PC <-> Node1 <-> Node2 <-> Node3

I’m planning the next steps to create PC software for monitoring and control and internal messaging infrastructure. While my comms topology is not ideal, that’s what I've got. The internal UART links are optical and cross 70KV isolation boundaries. This prompted (rightly or wrongly) a move away from the usual 'multidrop topology'

The ideal solution for my use case would an 'AMNB style' communication infrastructure - but suitable/adaptable for use with point-to-point connections. There is an element of message routing required.
I reviewed AMNB, more suited to a multidrop UART which this isn't. Maybe I can use it in a way that implements some sort of message forwarding within the handlers – but it just feels ‘at odds with’ the multidrop philosophy.

Reviewed xpcc. Ideal for CANBUS based comms.

sab / sab2 could be run in a polled mode  - but a certain amount of glue would be required for nodes to broadcasts and inter-node messages. Doable.

ROS has options for the managing of the HostPC to Node1 connection - but can't help with internal messaging. (AFAIK). I'd also need to embrace the roscore - which has to run somewhere, and that's quite a heavyweight. While interesting, it's not what I’m really after. I feel extending the internal protocol to the HostPC may be the best approach to begin with.

Lastly, there is 'RPR' from @chris-durand  - but without any examples, I'm having a hard time visualising how it works / how to interact with it. Are there any other sample projects of even non-functioning snippets that may give a clue as to how to build it into my application code.

Valid advice may also be - "Pull your finger out and go it alone".

Thanks all.

salkinium, 2021-07-03:
Yeah, I agree with your assessment and think a ring network might be useful for your setup.

The RPR which is a token ring network implementation, [there's a little more description in the old xpcc repo](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/communication/rpr.hpp). It was actually written by me in 2012 and I didn't port it to modm since it's not very well written, it has neither unit tests nor examples in xpcc. I mostly wrote it as an exercise with only an experimental setup. I have much more confidence in AMNB, since that's much better tested. 

I would throw it away and implement it from scratch with some inspiration from AMNB, although I'm not sure how much the Message format fits your routing requirements (the header is pretty small). Alternatively perhaps there are already stand-alone ring network comms out there? It's been a while since I checked for that topology…

jnewcomb, 2021-07-04:
I'll roll my own. Thanks for the advice.

closed 2021-07-04 by jnewcomb