modm-io/modm #650: Overflow in micro- / millisecond delay

chris-durand, 2021-07-06:
With the latest delay improvements the maximum permitted delay value for `delay_us`/`delay_ms` reduced to some extend. Right now `modm::delay_ms` overflows around 1500ms on an STM32F4 at 180 MHz. With the old implementation a delay of 5000ms worked just fine.

This is easily reproduced by adjusting the delay in the blink example and building it as debug. The assertion for overflow in `modm::delay_us` is constantly triggered and "modm::delay(us) has overflowed at the current frequency" is printed.

The problem was introduced in #642 but got worse with #648. I would regard this issue as somewhat critical since it silently breaks user code at runtime.

@salkinium

salkinium, 2021-07-06:
This is not a bug, the delay functions are only guaranteed to work at least 1s.

chris-durand, 2021-07-06:
Before the refactoring a comment in the source code promised 10s: "modm::delay(us) can only delay a maximum of ~10 seconds" so I assumed it should work.

chris-durand, 2021-07-06:
I also could not find this explicitly stated in the documentation, but maybe I am just blind.

salkinium, 2021-07-06:
Hm yes, I also checked and the 1s was never documented, that's pretty bad. I'll add a warning to the interface docs.

I checked the examples, and I'll also fix the 1.5s and 2s 

```
modm::delay(0.5s);
modm::delay(1.5s);
modm::delay(1000ms);
modm::delay(100ms);
modm::delay(100us);
modm::delay(10ms);
modm::delay(130us);
modm::delay(150ms);
modm::delay(180ms);
modm::delay(1ms);
modm::delay(1s);
modm::delay(200ms);
modm::delay(20ms);
modm::delay(250ms);
modm::delay(25us);
modm::delay(2ms);
modm::delay(2s);
modm::delay(2us);
modm::delay(300ms);
modm::delay(500ms);
modm::delay(50ms);
modm::delay(900ms);
modm::delay(Board::Button::read() ? 0.5s : 1s);
modm::delay(Board::Button::read() ? 125ms : 500ms);
modm::delay(Board::Button::read() ? 250ms : 500ms);
modm::delay(Button::read() ? 100ms : 300ms);
modm::delay(Button::read() ? 100ms : 500ms);
modm::delay(Button::read() ? 125ms : 500ms);
modm::delay(Button::read() ? 250ms : 1s);
modm::delay(Button::read() ? 250ms : 500ms);
modm::delay(Button::read() ? 500ms : 1s);
modm::delay(std::chrono::milliseconds(10*ii));
modm::delay_ms(500);
modm::delay_ms(Button::read() ? 100 : 500);
modm::delay_ms(data[ii]);
modm::delay_ns(ns);
modm::delay_us(1);
modm::delay_us(rx_settle);
modm::delay_us(us);
```

In modm/src/modm it's less of a problem:

```
modm::delay(100ns);
modm::delay(10us);
modm::delay(1us);
modm::delay(200us);
modm::delay(20us);
modm::delay(300ms);
modm::delay(500ms);
modm::delay(50us);
modm::delay(A);
modm::delay(B);
modm::delay(C);
modm::delay(D);
modm::delay(DATA_SET_UP_TIME);
modm::delay(DELAY_CE);
modm::delay(DELAY_CLK);
modm::delay(DELAY_TIME);
modm::delay(DELAY_TIME2);
modm::delay(E);
modm::delay(F);
modm::delay(G);
modm::delay(H);
modm::delay(I);
modm::delay(J);
modm::delay(MIN_E_HIGH_TIME);
modm::delay(MIN_E_LOW_TIME);
modm::delay(ns) can only delay a (positive) maximum of ~1 millisecond!");
modm::delay(us) has overflowed at the current frequency!");
modm::delay_ms(1);
modm::delay_ms(10);
modm::delay_ms(100);
modm::delay_ms(120);
modm::delay_ms(151);
modm::delay_ms(180);
modm::delay_ms(20);
modm::delay_ms(25);
modm::delay_ms(30);
modm::delay_ms(300);
modm::delay_ms(40);
modm::delay_ms(5);
modm::delay_ms(50);
modm::delay_ns(10);
modm::delay_ns(100);
modm::delay_ns(120);
modm::delay_ns(500);
modm::delay_ns(delayTime);
modm::delay_ns(delayTime*2);
modm::delay_us(1);
modm::delay_us(1);	// modm::delay_ns(100);
modm::delay_us(10);
modm::delay_us(100);
modm::delay_us(1200);
modm::delay_us(135);
modm::delay_us(15);
modm::delay_us(19);
modm::delay_us(2);
modm::delay_us(20);
modm::delay_us(200);
modm::delay_us(27);
modm::delay_us(30);
modm::delay_us(4);
modm::delay_us(5);
modm::delay_us(PHY::ResetDelay);
modm::delay_us(delay);
```

chris-durand, 2021-07-06:
Can't we implement millisecond delay by calling `delay_us` in a loop or something similar? For milliseconds the reduced accuracy by doing so should be quite fair.

salkinium, 2021-07-06:
Yes, I'm working on that right now, since this limitation is quite silly for (milli-)seconds.

PR #651 (2021-07-06): [core] Much longer modm::delay_ms on Cortex-M
There is a duration limitation due to accuracy in the cycle computation on `modm::delay_us` to at least 1s. This makes `modm::delay_ms` count ms and therefore has a 2^32/1000 seconds duration, which… uhm should be enough (49 days?).

- [x] Improve interface documentation
- [x] Fix delay_ns assertion?
- [x] Fix delay_ms limitations
- AVR already implements ms counting, so it's not limited.

Closes #650.

cc @chris-durand
closed 2021-07-06 by salkinium