modm-io/modm #653: modm:driver:mmc5603 recursive function call

SgtPepperFTW, 2021-07-07:
`		float inline
		swapDataFloat(uint8_t index) const
		{ return swapDataFloat(index) / 16384.f; }`

should be

`		float inline
		swapDataFloat(uint8_t index) const
		{ return swapData(index) / 16384.f; }`

salkinium, 2021-07-07:
Whoops, that's not right! 

Want to open a PR? It's super easy with the online editor: https://github.com/modm-io/modm/edit/develop/src/modm/driver/inertial/mmc5603.hpp#L249

salkinium, 2021-07-07:
Btw, how are you soldering the 0.8x0.8mm BGA package onto the board? I found that manually placing the thing on a hot plate works better than a reflow oven, since I can still change the position on the hot plate.

SgtPepperFTW, 2021-07-07:
Made a PR. 

Can't help with the soldering, I'm working with evaluation board in a breadboard, a device prototype is ordered from a factory.

salkinium, 2021-07-07:
Ahh, I don't haven enough volume for that.
I used them for rotation encoders [in mechanical model railway signals](https://salkinium.com/elva/signal_form.html), works great but requires some calibration.

[image]

closed 2021-07-07 by SgtPepperFTW

rleh, 2021-07-07:
> Ahh, I don't haven enough volume for that.

This can be done from a quantity of 1, I have had good experience with https://circuithub.com/ :wink: