modm-io/modm #655: modm:driver:mmc5603 converting 3 bytes for axis value into 32 bits variable

SgtPepperFTW, 2021-07-08:
The datasheet page looks like this
[image]

so uint8_t x0, that takes on its own bits [7:0], has to get, according to datasheet page, to bits [19:12] in a uint32_t, so we need to shift left (19-7) = 12 times. 
uint8_t x1, that takes on its own bits [7:0], has to get, according to datasheet page, to bits [11:4] in a uint32_t, so we need to shift left [11-7] = 4 times.
uint8_t x2, that takes on its own bits [7:0] has to be shifted right 4 times, cause 4 LSB are insignificant.

The lines

`swapData(uint8_t index) const
		{ return int32_t(((data[2*index] << 16) | (data[2*index+1] << 8) | (data[6+index])) >> 4) - 524288l; }`

should be

`swapData(uint8_t index) const
		{ return int32_t(((data[2*index] << 12) | (data[2*index+1] << 4) | (data[6+index])) >> 4) - 524288l; }`

Or?

chris-durand, 2021-07-08:
Did you see the whole expression is shifted down by 4 not only `xout2`?

The formatting also mislead me first trying to read it. Multiline version:
```
auto swapData(uint8_t index) const
{
    const auto xout0 = (data[2*index] << 16);
    const auto xout1 = (data[2*index+1] << 8);
    const auto xout2 = data[6+index];

    return int32_t((xout0 | xout1 | xout2) >> 4) - 524288l;
}
```
Thus, `xout0` is shifted left by 16 and then right by 4 which yields a shift left by 12 in total.

@salkinium This code will break on AVR, though. The type of `(data[2*index] << 16)` is `int` due to C++ integer promotion rules which is 16 bit on AVR. Thus, the shift computation will overflow. You would need to write `int32_t(data[2*index]) << 16` to make it work.

SgtPepperFTW, 2021-07-08:
Oh yeah, my bad.

closed 2021-07-08 by SgtPepperFTW