modm-io/modm #660: Rewrite GraphicDisplay

TomSaw, 2021-07-13:
Hello back! Long session passed, changed some lines of the Display-API to make graphics appear faster by factor.
This needs to be discussed, explained and cut in pieces, therefore the Issue.

Here's a little preview, running on STM32F0 @48Mhz

https://user-images.githubusercontent.com/738535/125512287-f0789d28-b91d-4732-8b4b-c6905b9cad61.mp4

The Number is rendered into a single monochrome Buffer witch is then streamed to different displays. Only the area of the Buffer is affected. Notice the noizy bottom half of the OLEDs. The flickering squares on the ILI9341 comes from debugging code. Soo yes, just a preview ;) Details about the implementation, the benefits and more examples coming soon.

**Let's get first rid of the obvious stuff: I'm gonna PR a lvl-up for ILI9341 in the next days.**

## Planned PRs

1. driver::Ili9341 Resumable Functions and optimizations
2. ui::color::RGB565 finetuning
3. Optimize most drawing functions
4. Seperate GraphicBuffer<Color, Width, Height> from GraphicDisplay<Color, Width, Height>

TomSaw, 2021-07-20:
Found another promising library not yet mentioned in modms GUI-Issues nor uPainter-Docs.
It also builds up on DMA2D and the docs contain some good [introduction to graphics on MCUs](https://support.touchgfx.com/docs/basic-concepts/embedded-graphics):

[TouchGFX](https://support.touchgfx.com)

chris-durand, 2021-07-20:
> If I remember correctly the TouchGFX license does not allow usage on non-ST devices and/or has expensive fees if used in commercial applications.

It's free even for commercial applications but the license only permits use with devices manufactured by ST.

TomSaw, 2021-07-20:
Ok, at least it's a good resource. Their [explanations](https://support.touchgfx.com/docs/basic-concepts/framebuffer) are very nice to read.

TomSaw, 2021-07-22:
Finally, my Workspace 🧮 and Neurons 🧠 are ready to get known to uPainter 🎨

**Few questions on the short @salkinium:**

- A: I've made a similar thing like the uPainters-`Painter` class, so a software or hardware painter can be plugged. Double work but was good for my understanding. Today i found uPainters-`Painter` is kinda reconstruction of QPainter, right?
- B: What's the current solution for fixed-point-arithmetics if there's any? CMSIS?

TomSaw, 2021-07-30:
First excerpt-PR for ILI9341  is out: #664
I've also PR-ed the whole code #665
👉 Please correct me if there's a better best-practice for such a situation

salkinium, 2021-07-30:
> What's the current solution for fixed-point-arithmetics if there's any? CMSIS?

There is none. There's a fixed point class in uPainter, however, it's a little outdated and not as flexible.
This seems to be the most promising: https://github.com/MikeLankamp/fpm

> Please correct me if there's a better practice for my situation of contributing this huge thing in chunks.

Just make one big PR and partition your changes into separate commits, since we can review the changes by commits instead of by files too (see the tab Commits).

TomSaw, 2021-07-30:
> Just make one big PR and partition your changes into separate commits, since we can review the changes by commits instead of by files too (see the tab Commits).

So then all changes will be in one branch 
? Makes it difficult to evolve the single changes.

salkinium, 2021-07-31:
> So then all changes will be in one branch? Makes it difficult to evolve the single changes, doesn't it?

Yes, you group them by commits and use fixup commits and interactive rebase to apply patches on you branch. This is a very typical git workflow and helps you break down large order-dependent changes.

You can even collaborate like this, if you carefully synchronize the rebase and force-pushes. We've done this recently for our STM32H7 work.

TomSaw, 2021-07-31:
Okay, thanks i'm going to learn and apply this.
Let's go for review of #665 anyways. I expect 'some' change requests so it's more efficient, splitting #665 more to the end of the workflow!

TomSaw, 2021-08-12:
I'm researching a board with DMA2D + Touch-TFT (ILI9341) to get my hands on the serious accelerators.

I could grab a [STM32F429 Discovery](https://www.st.com/en/evaluation-tools/32f429idiscovery.html) for small money but can't determine the type of the assemblied Display. Also the appropriate modm:board:disco-f429zi doesn't leave a note.

The're more Disco-boards with backpacked TFTs and weak technical details.
**How do you source this information?**

salkinium, 2021-08-12:
According to the [User Manual](https://www.st.com/resource/en/user_manual/um1670-discovery-kit-with-stm32f429zi-mcu-stmicroelectronics.pdf#page19), the F429 Disco has a TFT ILI9341 display. I would however recommend the F469 Disco with a MIPI-DSI panel with capacitive touch, even tough it's expensive (~50€).

closed 2021-09-29 by salkinium