modm-io/modm #663: ADC Undefined_Handler called

jnewcomb, 2021-07-28:
Hi Folks,
Just a quick sanity check please. STM32F429 (F4 discovery) ADC
There is a single global ADC interrupt on this device
[image]

Is modm incorrectly generating 3 separate named IRQHandlers?
Could this be why modm_abandon() reports Undefined_Handler exception when trying to use interrupts?
````
name = 0x08078570 "nvic.undef"
"An undefined NVIC interrupt was raised!"
context = 18
````
According to the Linker MAP file Cross Reference Table:
Symbol    
````
ADC1_IRQHandler                                   ../modm/CMakeFiles/modm.dir/src/modm/platform/adc/adc_interrupt_1.cpp.obj
ADC2_IRQHandler                                   ../modm/CMakeFiles/modm.dir/src/modm/platform/adc/adc_interrupt_2.cpp.obj
ADC3_IRQHandler                                   ../modm/CMakeFiles/modm.dir/src/modm/platform/adc/adc_interrupt_3.cpp.obj
ADC_IRQHandler                                    ../modm/CMakeFiles/modm.dir/src/modm/platform/core/vectors.c.obj
````

For the UARTs - Its as expected - the names match:
````
USART1_IRQHandler                                 ../modm/CMakeFiles/modm.dir/src/modm/platform/uart/uart_1.cpp.obj
                                                  ../modm/CMakeFiles/modm.dir/src/modm/platform/core/vectors.c.obj
USART2_IRQHandler                                 ../modm/CMakeFiles/modm.dir/src/modm/platform/uart/uart_2.cpp.obj
                                                  ../modm/CMakeFiles/modm.dir/src/modm/platform/core/vectors.c.obj
USART3_IRQHandler                                 ../modm/CMakeFiles/modm.dir/src/modm/platform/uart/uart_3.cpp.obj
                                                  ../modm/CMakeFiles/modm.dir/src/modm/platform/core/vectors.c.obj
USART6_IRQHandler                                 ../modm/CMakeFiles/modm.dir/src/modm/platform/uart/uart_6.cpp.obj
                                                  ../modm/CMakeFiles/modm.dir/src/modm/platform/core/vectors.c.obj
````

salkinium, 2021-07-28:
> Is modm incorrectly generating 3 separate named IRQHandlers?

Yes. The [heuristic here is wrong](https://github.com/modm-io/modm/blob/develop/src/modm/platform/adc/stm32/module.lb#L78-L90): The F4 is wrongly presumed to not have shared ADC IRQs. 

This problem of shared IRQs between instances and peripherals is coming up a lot, we need to fix that in a common way, but for your problem you should hack around those lines to fix it. Pretty sure the F429 is not the only one with a shared ADC IRQ.

salkinium, 2021-07-28:
Just debugged this on the F4 discover ADC example and the issue is in line 79 (and a later failing comparison between string and int): 

```diff
-         props["shared_irqs"] = {"ADC": listify(device.get_driver("adc")["instance"])}
+         props["shared_irqs"] = {"ADC": list(map(int, listify(device.get_driver("adc")["instance"])))}
```

jnewcomb, 2021-07-28:
I wasn't even aware of adc_shared_interrupts.cpp.
Applied the patch and the files generate as expected. Many thanks.
..
The only issue I'm facing is that interrupts don't get cleared.. when I drop out the interrupt_hander() I go straight back into MODM_ISR(ADC)

I've tried acknowledgeInterruptFlags for ADC1 and ADC2; will keep digging..

````
  void Enable()
  {
    Adc3::initialize<Board::SystemClock>();
    Adc3::connect< TEMPMOD_AdcIn4::In4>();
    Adc3::setPinChannel<TEMPMOD_AdcIn4>(Adc3::SampleTime::Cycles480);

    Adc3::enableInterruptVector(5);
    Adc3::enableInterrupt(Adc3::Interrupt::EndOfRegularConversion);

    AdcInterrupt3::attachInterruptHandler(interrupt_handler);
  }

  void StartConversion()
  {
    Adc3::startConversion();
  }

  static void interrupt_handler()
  {
    Adc3::acknowledgeInterruptFlags(Adc3::InterruptFlag::All);
  }
````

salkinium, 2021-07-28:
There is a bug in the `acknowledgeInterruptFlags`. The ADC3->SR register is marked `rc_w0`, which means "Software can read as well as clear this bit by writing 0. Writing ‘1’ has no effect on the bit value.". Unfortunately [the code for this writes 1, instead of 0](https://github.com/modm-io/modm/blob/develop/src/modm/platform/adc/stm32/adc_impl.hpp.in#L250-L256):

```diff
void
modm::platform::Adc{{ id }}::acknowledgeInterruptFlags(const InterruptFlag_t flags)
{
- 	// Flags are cleared by writing a one to the flag position.
- 	// Writing a zero is ignored.
- 	ADC{{ per }}->SR = flags.value;
+ 	ADC{{ per }}->SR &= ~flags.value;
}
```

Since this is the generic ADC driver, perhaps there is a difference between the remaining STM32 families, which makes this change wrong again for those families. Gotta check that later.

jnewcomb, 2021-07-28:
Well spotted! That's got me going again...
A quick glance at the reference manual hints this is also a problem for the F1 and F2 series.

salkinium, 2021-07-28:
Awesome, want to open a PR with all the patches and a revised implementation for F1/F2/F4? I'm suspicious now, there are probably more issues with this driver.

closed 2021-09-17 by salkinium