modm-io/modm #666: Teach DMA to send n bytes n times

TomSaw, 2021-07-30:
To accelerate the graphic APIs `clear(Color color)` methods i need a DMA feature.
See #665

#### Application 1: clear external CGRAM (Display-RAM) with DMA:

`// This may be the method to implement
SpiMaster1_Dma::transfer(const uint8_t *tx, uint8_t *rx, std::size_t length, uint32_t repeat);`

`// Example to send: 0xAA, 0xBB, 0xAA, 0xBB, 0xAA, 0xBB, 0xAA, 0xBB, 0xAA, 0xBB, 0xAA, 0xBB, ... 128 repeats
uint8_t buffer[2] = {0xAA, 0xBB};
SpiMaster1_Dma::transfer(buffer, nullptr, 2, 128);`

#### Application 2: Substitute std::fill(..)
Similar interface like for Application 1

## Solution
Sending repeative bytes is easy: Just disable the Address-incement. For a multi-byte sequence i think we can use STMs [Circular Mode](https://community.st.com/s/question/0D53W00000IXdAuSAL/dma-normal-vs-circular).
Whats your guess @chris-durand / @mikewolfram / @salkinium ? Seen you've put effort into the DMA soon.

mikewolfram, 2021-08-04:
I had a quick look into the L4 reference manual. If I interpret the SPI chapter correctly, then it should be possible to write a 16bit value to an 8bit SPI (DMA packing on SPI, pages 1316 and 1318). With pointer incrementing disabled this would allow to set the pixel color for a predefined area. But requires reconfiguration of the DMA channel before sending.

The problem with circular mode is the never ending transfer. You'd need to count the loops and disable the transfer when done. If going that path it would be preferable to use a small buffer for the transfer to reduce the overhead of the DMA IRQ handler being called after every byte/word.

TomSaw, 2021-08-06:
Ah nice! Thanks for the hint mike, I'll gonna try this out.

Additional note:
For the graphic usecase, the highest expected value for repeat is **3**
Maximum color depth used for simple displays: 24bpp -> **3** bytes.

salkinium, 2021-08-06:
At least for STM32 DMA you can make it read 16- or 32-bits (but not 24-bits), but only write 8-bits with the DMA buffering the value in between, so this may work even without anything special.

I would however, focus on DMA2D first, since replacing parts of libc(++) isn't trivial and the overhead of setting up DMA for small buffers is bigger than memset/memcpy. You also need a future/promise mechanism so you can do useful CPU work while you wait for DMA to finish.

TomSaw, 2021-08-09:
Si.  Let's keep std::fill, it's fine.

The OLEDs actually don't have an internal buffer. If I can't implement a 'clear()' working this way, I need to add a (little) buffer just for  clearing the noize on power-up witch is stupid.

TomSaw, 2021-09-13:
I have expanded this task and added bit-resolution to each `transfer<4-16>(...)` of SPI. The desired feature `Spi::transfer(data, repeat)` works and speeds up clearing and drawing simple shapes. One little bug left but should be easy...

  While playing around, another accelerator has unveiled: 👉 **Hardware endianess** 👈 for for data between 9-16Bits, transfered via SPI. All of the `modm::fromBigEndian(...)` sanities (in Ili9341, but there's more) can be spared.

Have stacked this into #665. It's a round package and could be PRed earlier if you want.

#### Transfer 4x 6bits via Spi
```
uint8_t data[] = {1,2,3,4};
RF_CALL(SpiMaster::transfer<6>(data, nullptr, 4));
```
[image]

### Hardware endianess for Spi transfer
```
union {
	uint8_t data8[2] = {0xDE, 0xAD};
	uint16_t data16;
};
RF_CALL(SpiMaster::transfer<8>(data8, nullptr, 2)); // outputs 0xDEAD
RF_CALL(SpiMaster::transfer<16>(data16)); // outputs 0xADDE
```
##### Without DMA
[image]

##### With DMA
[image]

TomSaw, 2021-09-14:
Oh man 🤩 this 16bit `Spi::transfer(...)` really is a missing piece. Look on this improved `Touch2046::getRawValues()`:
### Before
```
template < class SpiMaster, class Cs >
modm::ResumableResult<std::tuple<uint16_t,uint16_t,uint16_t>>
modm::Touch2046<SpiMaster, Cs>::getRawValues()
{
	RF_BEGIN();

	RF_WAIT_UNTIL(this->acquireMaster());
	Cs::reset();

	RF_CALL(SpiMaster::transfer(
		bufferWrite.data(),
		reinterpret_cast<uint8_t*>(bufferRead.data()) + 1,
		17));

	z = 4095 + (modm::fromBigEndian(bufferRead[1]) >> 3)
		- (modm::fromBigEndian(bufferRead[2]) >> 3);

	y = (modm::fromBigEndian(bufferRead[3]) >> 3)
		+ (modm::fromBigEndian(bufferRead[5]) >> 3)
		+ (modm::fromBigEndian(bufferRead[7]) >> 3);

	x = (modm::fromBigEndian(bufferRead[4]) >> 3)
		+ (modm::fromBigEndian(bufferRead[6]) >> 3)
		+ (modm::fromBigEndian(bufferRead[8]) >> 3);

	if (this->releaseMaster()) {
		Cs::set();
	}

	RF_END_RETURN(std::make_tuple(x, y, z));
}
```
### After
```
template<class SpiMaster, class Cs, Resolution R>
modm::ResumableResult<std::tuple<uint16_t, uint16_t, uint16_t>>
modm::Touch2046<SpiMaster, Cs, R>::getRawValues()
{
	RF_BEGIN();

	RF_WAIT_UNTIL(this->acquireMaster());
	Cs::reset();

	RF_CALL(SpiMaster::template transfer<16>(bufferWrite.data(), bufferRead.data(), 8));

	z = 4095 + (bufferRead[1] >> 3) - (bufferRead[2] >> 3);
	y = (bufferRead[3] >> 3) + (bufferRead[5] >> 3) + (bufferRead[7] >> 3);
	x = (bufferRead[4] >> 3) + (bufferRead[6] >> 3) + (bufferRead[8] >> 3);

	if (this->releaseMaster())
		Cs::set();

	RF_END_RETURN(std::make_tuple(x, y, z));
}
```

closed 2021-09-28 by TomSaw