modm-io/modm #670: Support grayscale displays

twasilczyk, 2021-08-20:
modm::ui::display either supports monochrome or color displays. The two displays I've been working with (SSD1322 and ST7586S) are 16- and 4-level grayscale.

One option would be to make ColorGraphicDisplay a template allowing other color profiles than Rgb565 + add 16/4 level color profile. Then maybe even MonochromeGraphicDisplay could just use the same template with 1-bit color profile. There's a challenge though: in 16-level grayscale displays, two pixels are encoded in a single byte.

While we're at modm::ui::display, there's also one benefit of using these controllers that modm doesn't utilize: you can select a fragment of the display and not update the whole thing.

salkinium, 2021-08-21:
@TomSaw is our display expert now, his https://github.com/modm-io/modm/pull/665 is addressing some of these issues.

TomSaw, 2021-08-21:
Yepp I've also had my concerns with the display implementations in the beginning of this year.
The #665 rewrite considers any type of color including grayscale of any depth.

However, not all cases **where multiple pixels are stuffed into one byte** are covered yet:

- `modm::color::Grayscale<>` needs implementation for n-bit grayscale-depths to cover f.e. 4 and 16 level Grayscale.

- `modm::graphic::Buffer` needs according implementation.

I've added this to the TODOs of #665. Think, I've got some SSD1322 around so I can test this. Stay tuned!

twasilczyk, 2021-08-21:
This is awesome and I'm glad to hear you're considering multi-pixel-per-byte cases. There's one extra complication with SSD1322 that you need to be aware of: addressing is per word, not per byte. So a single 16-bit cell in memory represents four 4-bit pixels - this affects the widths of partial drawing sprites.

Did you consider extending support to drawing just parts of the display? This speeds up most of the use cases *a lot*. In my own project I easily hit north of 100 FPS (that's where my counter ends) even with a very inefficient SPI driver. No wonder why - each frame was sending just a handful of bytes. Added benefit is that you don't need RAM - you can drive a large OLED with something as small as atmega8.

TomSaw, 2021-08-23:
`a single 16-bit cell in memory represents four 4-bit pixels - this affects the widths of partial drawing sprites.`
Ok, good to know! This should only affect the transfer-algorithm of the display-driver. I'll keep this in mind when debugging ;).

`Did you consider extending support to drawing just parts of the display`
That's already in the #665 Box 😋. You can define arbitrary graphic-buffer with arbitrary dimensions and color-type and send it to any other arbitrary graphic-buffer or display.

- One can fragment the display into several areas and update each area asynchroniously.
- Another interesting scenario is writing a buffer with low color-depth (f.e. 2-bit/pixel) to display with higher color-depth. The display-driver converts the buffer-pixels to it's predefined `color` on demand using a small local conversion-buffer. This spares a ton of RAM when f.e. writing just text where no muti-colors are required.

**Spend some time today on the generic colors / buffers and will add some self-explaining examples for the new APIs to #665 as soon as the generic colors and buffers are good enough...**

## ... In short, this is the new workflow:

### 1. Declare your color-type
`using Mono = GrayscaleT<1>; // Monochrome 1 bit/pixel`
`using Gray2 = GrayscaleT<2>; // Grayscale 2 bits/pixel`
`using Gray4 = GrayscaleT<4>; // Grayscale: 4 bits/pixel`
`using Gray3 = GrayscaleT<3>; // Grayscale: 3 bits/pixel for whatever phantasy application`
`using Rgb565 = RgbT<5,6,5>; // Rgb565-Color: 5bits red, 6bits green, 5bits blue/pixel`
`using Rgb888 = RgbT<8,8,8>; // Rgb888-Color: 8bits per red, green,blue / pixel`
`using Hsv888 = HsvT<8,8,8>; // Hsv888-Color: 8bits per hue, sat, value / pixel`

### 2. Declare your buffer-type
When using `Mono`, `Gray2` or `Gray4` as color-type, the buffers internal data-scheme already stacks the pixels, hence memory is saved  and can be transfered to the display without additional conversion.

`using MyBuffer = modm::Buffer<Gray16, 100, 123> // Graphic-buffer with x=100, y=123`

### 3. Draw and write into the buffer (similar like before)
`MyBuffer.draw(Line(...));`
`MyBuffer.draw(Circle(...));`
`MyBuffer << "hello Buffer";`
`MyBuffer.write(AnotherBuffer, {20, 30});`

### 4. Finally send the buffer to a display with optional position. When the buffer overflows the display, it's clipped. Therefore you can move buffers around on the display to create nice animations. 
`display.write(MyBuffer); // Position defaults to {0, 0}`
`display.write(MyBuffer, {-12, 34}); // specific position`

twasilczyk, 2021-08-24:
This is awesome, exactly (or more than) what I expected from embedded display library.

In my own personal project I did one more thing with buffers and embedded images. Writable `Buffer` inherits from read-only `Image`, so that you can use the same code to display RAM-backed buffers and Flash-backed images (with no ram usage). And I have a Python converter script from png to header files that also embeds all the metadata like size.

twasilczyk, 2021-08-28:
I can’t promise anything, since modm image APIs are just a prerequisite to a side project (Smart Response XE BSP) to my other hobby project that I temporarily gave a break :D.

Having said that, a converter script with png and multi-color mode support shouldn’t be far from what I currently have. I’ll try to give it a shot.

TomSaw, 2021-08-29:
No problem, just had the feeling you're willing to bring yourself in.
For me there's no prio so don't worry and have fun with programming!

TomSaw, 2021-09-03:
#665 now supports stacked monochrome, gray1, gray2 and gray4 Buffers.

I have slept a night over the ST7586 pixel-format you've mentioned in #673 - where one byte just contains 3 2bit-pixels - and came out with this:

#### Facts
- Buffer-manipulation (drawing shapes, copying etc.) is more expensive than buffer-conversion.
- Algorithms working on symetric patterns optimize much better than for fuzzy patterns. Looking at you ST7586-pixelformat 👀
- One does not want to waste 2bits in every byte of a potentially big buffer.

#### Outcome
1. A regular generic `modm::Buffer<Gray2>, Resolution>` should be used. 
2. Implement the `modm::St7586::write(BufferInterface<Gray2> &buffer)` with integrated conversion!
You can convert pixel by pixel but i absolutely recommend a small local conversion-buffer and do it in chunks. That's also a must have for **Direct memory access**.

Please check `Ili9341::writeImage(Decoder<Monochrome, Accessor>, Point)` in #665 witch is very similar. It converts a Monochrome buffer to Rgb565-color on demand.

From my side, issue can be closed.

twasilczyk, 2021-09-04:
In general, I agree, but there's one important counter-point: in its current form, pixel conversion from monochrome to st7586s monochrome format takes 3x more time than transmission itself (checked with logic analyzer, but I don't remember exact numbers), what brings FPS down significantly. If the implementation supported both Gray2 and Gray332 formats, one could pick former for fast buffer manipulation and the latter for fast drawing directly from flash.

Having said that, I'm happy with the performance with on-the-fly conversion.

For closing this (or any other) issue, I would wait for merging the PR. Up to you.

TomSaw, 2021-09-06:
Buffers purpose is to enable random access of pixels. A symetric pixel-layout will outperform anything else when drawing lines and circles. 

Simple copies of images from flash to display **should not involve any buffer nor the color-classes**: It's the Display drivers business who takes a `modm::accessor::Flash` into account.

I've improved these functionalities for Ili9341, Ssd1306 and Sh1106 in #665 as well but have not added further color-models: just the good old monochrome for now. The key to make modm Gray332-ready is to improve the image conversion script. Afterwards some reusable supportive classes around `modm::accessor::*` would be nice to grab the metadata.

True, issue is not solved. Let's wait for #665  and keep it open until things run as expected.