modm-io/modm #672: Add Integer Type Selection for generic programming

TomSaw, 2021-08-24:
I quite often need a boost like [Integer Type Selection](https://www.boost.org/doc/libs/1_47_0/libs/integer/doc/html/boost_integer/integer.html).

Shall i port this or is there a another plan in the shelf?

TomSaw, 2021-08-24:
Something like this:

template <unsigned int Bits>
struct smallest_integer
{
	using type = std::conditional_t<
		(Bits <= 8), uint8_t, std::conditional_t<
		(Bits <= 16), uint16_t, std::conditional_t<
		(Bits <= 32), uint32_t, std::enable_if_t<
		(Bits <= 64), uint64_t>
	>>>;
};

template <unsigned int Bits>
using smallest_integer_t = typename smallest_integer<Bits>::type;

salkinium, 2021-08-24:
Not sure if I'm ready to integrate boost, so I'd port it.

TomSaw, 2021-08-30:
0kay i've kept it simple and added `/modm/math/utils/integer_traits.hpp` to #665 . It includes additional magic i needed for fully generic `modm:ui:color`.

closed 2021-08-30 by TomSaw