modm-io/modm #687: Floating point exception in __modm_startup

mcbridejc, 2021-09-17:
For no apparent reason, my project started failing to run today. 

When I attached debugger and started narrowing things down, I found that it got as far as `table_copy` in __modm_startup, and then an exception occurs and I land in UndefinedHandler.

Looking at UFSR, I see that the NOCP bit is set:Indicates that a Cortex-M coprocessor instruction was issued but the coprocessor was disabled or not present.

Sure enough, the last instruction was a `vldr`. It appears that GCC is using the floating point register as temporary storage, or something along those lines, during table_copy.

Moving the enable FPU line ahead of table_copy() in __modm_startup() fixed it, because the FPU was on when it hit the vldr. 

Apparently, there's also a `general-regs-only` attribute (https://gcc.gnu.org/onlinedocs/gcc-9.1.0/gcc/ARM-Function-Attributes.html), which can be put on a function to prevent these. Adding `#pragma GCC target ("general-regs-only")` in `startup.c.in` also fixed it. 

Either of those options should be a fine fix. It seems reasonable that GCC might do this; the file is compiled with hard floating point. However, I have no idea how I managed to make it happen when it wasn't already. 

I am building with cmake. When it suddenly failed I was attempting to override the tinyusb weak descriptor symbols, and having problems because the overriding symbols were being linked in from a static library, and apparently that doesn't necessarily work. I don't see why anything I did should have mattered, except perhaps that there's some particular table values that cause gcc to emit the floating point? 

I think the FPU enable should move ahead of the table copy, or perhaps even into Reset_Handler so it can be turned on before modm_initialize_platform?

mcbridejc, 2021-09-17:
`__modm_startup` side by side disassembly: with 'general-regs-only` pragma on left, without it on the right. 
The vldr is at 0x0801706a.

[image]

salkinium, 2021-09-17:
> into Reset_Handler so it can be turned on before modm_initialize_platform?

Yes, that's really the only choice, otherwise we would have to mark every function call before the enable with the pragma. Would be nice if we could do it as assembly as well, so we can be sure the compiler won't use FPU regs to enable the FPU regs (you never know…).

salkinium, 2021-09-17:
These three instructions work for me in `reset_handler.sx`:

```
ldr r0,=0xe000ed88
ldr r1,=0x00f00000
str r1, [r0, #0]
```

mcbridejc, 2021-09-17:
Ah cool. that's pretty much what I was about to commit, but I created a __modm_enable_fpu function, and used the general-regs-only pragma to ensure it couldn't emit the floating point. Mostly b/c I didn't want to write more than a bl instruction :).

salkinium, 2021-09-17:
Oh, sorry. It's been nagging me to add a comment about why we set the stack pointer and I've just been waiting for an excuse to touch that file again for a "real reason", so I couldn't pass this opportunity up…

PR #688 (2021-09-17): [cortex-m] Enable FPU before startup script
This allows the compiler to emit instructions that access the FPU
registers for copying.

Fixes #687.

cc @mcbridejc
closed 2021-09-17 by salkinium