modm-io/modm #691: ublox ZED-F9P Differential GNSS Module

kikass13, 2021-09-28:
Hello,

We have implemented a rs232 driver for the ublox [ZED-F9P](https://www.u-blox.com/en/product/zed-f9p-module?lang=de) differential gnss module. 

The source code is here:
 * https://github.com/DroidDrive/ublox_modm_example

The zed-f9p needs proper configuration. I have prepared this repository (utilizing wine & docker), so that a user can configure the ublox module properly (configure the UBX messages for example):
* https://github.com/DroidDrive/ublox_gnss_testbench

The specific breakout board we are currently using is [this](SparkFun GPS-RTK-SMA Breakout - ZED-F9P (Qwiic))

## general
The driver is very simple and has some problems, but it works fine.
Currently, these UBX messages are implemented:

 * UBX-NAV-STATUS
    * contains rtk information and general status flags
 *  UBX-NAV-CLOCK
     * contains clock and time information (drift, etc)
 * UBX-NAV-PVT
     * contains a lot of information regarding current geolocation (lat, lon) and  other stuff liek speed, heading, altitude etc.
 * UBX-NAV-COV
     * contains covariance matrix for NORTH / EAST / DOWN position and velocity

## what does it do

The driver simply polls for a byte and pushes it into the driver. The driver will receive data until a full frame was received. The Driver will then perform CRC checks and validate the frame. If the frame is valid, the data is parsed (put into a struct) and given to the user via callback.

## current problems
 * maybe someone can tell me, if a larger uart buffer is possible 
 *  and how I can transfer uart data via dma & interrupt in modm.

salkinium, 2021-09-28:
> maybe someone can tell me, if a larger uart buffer is possible

There is an lbuild option for buffer.tx and buffer.rx sizes for all UARTs. Tops out at 64kB, although I think that's just a software limitation. (Although I would assume that 64kB per direction is enough?).

> and how I can transfer uart data via dma & interrupt in modm.

Sorry, it's not implemented yet, although I would assume that you can orient yourself on the DMA SPI Master if you want to implement it yourself.

FYI: I also did [a project with some UBX parsing](https://github.com/salkinium/navimet) and found these [comms libraries to be quite useful](https://github.com/commschamp/cc.ublox.commsdsl). Not sure if that applies to your project though.

kikass13, 2021-09-28:
> There is an lbuild option for buffer.tx and buffer.rx sizes for all UARTs. Tops out at 64kB, although I think that's just a software limitation. (Although I would assume that 64kB per direction is enough?). 

is this documented somewhere? 

> Sorry, it's not implemented yet, although I would assume that you can orient yourself on the DMA SPI Master if you want to implement it yourself.

well thats tough luck i suppose .. :(

>FYI: I also did a project with some UBX parsing and found these comms libraries to be quite useful. Not sure if that applies to your project though.

yeah i saw that. but i did not want to bother with code generation for simple structs (or dynamic parsing).  I just wanted to implement a small solution without the overhead.

rleh, 2021-09-28:
https://modm.io/reference/module/modm-platform-uart-lpuart1/

salkinium, 2021-09-28:
> is this documented somewhere?

Yes, `lbuild discover -n "::uart:*"` and ideally also on the homepage, but apparently not on modm.io since I excluded the peripheral instance modules to cut down on the number of modules. Hmpf, the issues never stop… 😒

closed 2021-09-29 by salkinium