modm-io/modm #784: Debug python script

jasa, 2021-11-17:
On a linux (Fedora) platform it's not possible to press CTRL-C in GDB to stop execution without GDB terminates completely. I have solved the problem by adding a signal handler to `modm_tools/gdb.py` which does nothing except catching the signal. Does anyone have the same problem, on google it's a known problem then using gdb/openocd, I will be happy to make a PR since it's a annoying feature.

Solution:
``` python
import signal

def signal_handler(sig, frame):
    pass
```
And somewhere in main activate it:
``` python
signal.signal(signal.SIGINT, signal_handler)
```

salkinium, 2021-11-18:
> Does anyone have the same problem

Not me on macOS, but maybe @rleh?

> I will be happy to make a PR 

Yes, go for it, I'll test it on macOS!

jasa, 2021-11-18:
> Not me on macOS, but maybe @rleh?
Hmm, maybe a strange macOS behavior. 
I think it's common behavior in posix, the child process passes SIGINT to the master process (which it should or it would loose control) and if the master don't have a signal handler every thing dies.

salkinium, 2021-11-18:
It may also be related to how the Python wrapper passes the Ctrl-C to OpenOCD in the background process: https://github.com/modm-io/modm/blob/develop/tools/modm_tools/openocd.py#L92-L97

salkinium, 2021-11-18:
I'd be happy if you found a solution for this, perhaps this has been improved now in OpenOCD v0.11 (this was implemented with v0.10).

rleh, 2021-11-18:
> > Does anyone have the same problem
> 
> Not me on macOS, but maybe @rleh?

I also don't have the problem you described on my Fedora (34, with KDE) system using _Konsole_ or _Yakuake_ terminal.
I rember some weird behavior in the past when running `scons debug` in some other (broken) terminal emulators like the Vscode terminal or similar.

@jasa Could you provide more information about your environment and a little tutorial on how I can reproduce the problem?

jasa, 2021-11-18:
@rleh I have a default Fedora 34 with GNOME with the newest arm-none-eabi toolchain from developer.arm.com. I just don't work out of the box on 2 different computers. I'm not the only one with that problem, maybe it's related to the GNOME terminal?

jasa, 2021-11-23:
@salkinium  Is there any special reason why openocd is started from a script instead of gdb starts it directly?

 eg. `arm-none-eabi-gdb -iex 'target extended | openocd -c \"gdb_port pipe\" -f openocd.cfg'  -x gdbinit <ELF>`

salkinium, 2021-11-23:
Yes, I think there was an issue [with Ctrl-C in GDB ending the OpenOCD process in the background](https://github.com/modm-io/modm/blob/develop/tools/modm_tools/openocd.py#L92-L93).
It's been a few years since I tested this though, so perhaps this is fixed now?

jasa, 2021-11-29:
> Yes, I think there was an issue [with Ctrl-C in GDB ending the OpenOCD process in the background](https://github.com

@salkinium Yes it works, but not on Windows.
Then the `gdb.py` passes control to the child process `gdb` it could be considered a weak design if `gdb.py` exits on a SIGINT, a SIGTERM kills it anyway.  [Signal doc](https://www.baeldung.com/linux/sigint-and-other-termination-signals)
I'm now running openocd remotely on a raspberry pi so I'm out of this design and won't complain if you close it.

closed 2021-12-09 by salkinium