modm-io/modm #789: RTT on Blue Pill

StevenMacias, 2021-12-04:
Hi, I am currently trying to get RTT working on a blue pill. I read this issue https://github.com/modm-io/modm/issues/501 and as far as I understood the STLink-v2 (3v3, GND, SWDIO, SWCLK) should be enough to test this functionality. 

I did the following: 
```
# Copy the rtt example from stm32f3_discovery into the blue_pill_f103 examples
cd modm/examples
cp -r stm32f3_discovery/rtt blue_pill_f103/
# Copy the openocd.cfg since the blue pill doesn't have built-in programmer 
cp blue_pill_f103/blink/openocd.cfg blue_pill_f103/rtt 
```
Then I replace the `LedNorth` by `LedGreen` in the `blue_pill_f103/rtt/main.cpp`

Finally, edit the `blue_pill_f103/rtt/project.xml` by updating the `<extends>` field with the blue pill target and by adding the openocd.cfg option. It looks like this: 

```
<library>
  <extends>modm:blue-pill-f103</extends>
  <options>
    <option name="modm:build:build.path">../../../build/blue_pill_f103/rtt</option>
    <option name="modm:platform:rtt:buffer.rx">16</option>
    <option name="modm:build:openocd.cfg">openocd.cfg</option>
  </options>
  <modules>
    <module>modm:platform:rtt</module>
    <module>modm:processing:timer</module>
    <module>modm:build:scons</module>
    <module>modm:debug</module>
  </modules>
</library>

```
Now, I call: 
- `cd blue_pill_f103/rtt`
- `lbuild build`
- `scons -j8`
- `scons program profile=debug`
- `scons log-rtt`

This is how the final result looks like, no errors but no loop debug traces:
```
(base)  steven@legion  ~/workspace/steven/modm/examples/blue_pill_f103/rtt   develop  scons log-rtt
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
╭───OpenOCD───> Real Time Transfer
╰─────RTT────── stm32f103c8t6
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J17S4 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.281698
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : rtt: Searching for control block 'modm.rtt.modm'
Info : rtt: Control block found at 0x20000194
Info : Listening on port 9090 for rtt connections
```
What I am missing? Thanks in advance!

salkinium, 2021-12-05:
I can reproduce the problem: The issue is that the identifier is found at the wrong address, for me `0x200003c0`, but the `rtt_control` structure is at `0x20000c00`. If you unpower the F103 after programming it works fine, but if you program it again, you'll find the same issue. 

The issue is that the OpenOCD uploader uses the SRAM as a buffer to make flashing faster, thus it copies part of the firmware into SRAM and then does not erase it. This is a very nice Heisenbug, since it only shows if the `rtt_control` is near the end of the firmware in the last buffer that OpenOCD copies.

This is something I never considered, but we can "fix" it by zeroing the main SRAM after flashing. See if #790 fixes this for you.

StevenMacias, 2021-12-05:
Exactly, by power cycling the blue pill it works! Thanks for the help and I will check https://github.com/modm-io/modm/pull/790.

PR #790 (2021-12-07): [rtt] Zero SRAM when using RTT
This prevents OpenOCD from leaking RTT identifier into SRAM which is
used to buffer the firmware during uploading.

What a stupid bug… 🙄

Fixes #789.
closed 2021-12-07 by salkinium