modm-io/modm #793: hidden symbol `__dso_handle' isn't defined

kikass13, 2021-12-14:
modm commit https://github.com/modm-io/modm/commit/c7eadef7b034af86f0b187b6aeb5fa4d561d262b removed some stuff, which in turn leads to linker errors in one of my libraries. I am not exactly sure how to resolve this problem 

```
/opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libstdc++.a(system_error.o): in function `_GLOBAL__sub_I__ZSt20__throw_system_errori':
system_error.cc:(.text.startup._GLOBAL__sub_I__ZSt20__throw_system_errori+0x1c): undefined reference to `__dso_handle'
/opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libstdc++.a(cxx11-ios_failure.o): in function `(anonymous namespace)::__io_category_instance()':
cxx11-ios_failure.cc:(.text._ZN12_GLOBAL__N_122__io_category_instanceEv+0x38): undefined reference to `__dso_handle'
/opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc.a(lib_a-fini.o): in function `__libc_fini_array':
fini.c:(.text.__libc_fini_array+0x20): undefined reference to `_fini'
/opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: build/app/scons-release/app.elf: hidden symbol `__dso_handle' isn't defined
/opt/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: final link failed: bad value
```

I cannot really figure out where exactly this problem occurs,
maybe someone can help me :)

rleh, 2021-12-14:
Looks like your library was and is broken and calls something like `abort()` or `exit()` somewhere.

salkinium, 2021-12-14:
`__dso_handle` is used during execution of global destructors via `_fini` array, which we don't implement, since it makes no sense anyways (we just reboot).
Perhaps you're using some unported part of the std lib like `std::cout`?

chris-durand, 2021-12-14:
I suspect your code has an `#include <iostream>` somewhere. That header [instantiates a global object](https://github.com/gcc-mirror/gcc/blob/b0e6a257f1862e217cdf19332ea0f7bad56dcddc/libstdc%2B%2B-v3/include/std/iostream#L74) to allocate resources for input / output streams (`cout`, etc).

The non-template parts of the standard library ship as a binary with the compiler and are not compiled with `-fno-use-cxa-atexit`. Thus, destructors of objects with static storage duration defined in those binaries will require `__dso_handle`.

If you remove the problematic include everything should be fine. There is no support for iostreams anyway.

kikass13, 2021-12-14:
Thanks for the reply, sadly I did check for these things beforehand and could not solve the problem.

 * no <iostream> 
 * no exit() / abort()

I removed these prior to this post. I also added the dso_handle back but that still leaves me with the 
`fini.c:(.text.__libc_fini_array+0x20): undefined reference to `_fini'` error ...

I guess I have to use the older modm version for now and try to find the code responsible for my problem :)
thanks for the hints anyways <3

closed 2021-12-14 by kikass13

salkinium, 2021-12-14:
The linkerscript is responsible for discarding the `.fini` section. But note that global destructors were never really supported by modm and even if they are linked into the firmware we never call them.
https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/linker.macros#L232-L236

chris-durand, 2021-12-14:
Are you using https://github.com/DroidDrive/lely_modm in your project? If yes, that's where I would look first. The code of this lely library looks highly suspicious to me, just looking at some includes:
```
#include <iomanip>
#include <sstream>
#include <string>
#include <system_error>
```
Your linker errors suggest some code path could throw `std::system_error` and the system error category object for io streams is somehow pulled in as well (symbol `__io_category_instance` from `cxx11-ios_failure.cc` in libstdc++).

daixtrose, 2022-01-26:
> `__dso_handle` is used during execution of global destructors via `_fini` array, which we don't implement, since it makes no sense anyways (we just reboot).

But is it possible to have dummy symbols for these then? I mean better a no-op function than a linker error - especially when there is no harm.

kikass13, 2022-01-26:
> But is it possible to have dummy symbols for these then? I mean better a no-op function than a linker error - especially when there is no harm.

I have added the symbol to my code for now ... So this problem is done for me (:

rleh, 2022-01-26:
No-op would be an assertion resulting in a jump to a (non recoverable) fault handler, like the previous behavior.

> especially when there is no harm

There probably is a lot of harm.