modm-io/modm #808: Improve "thisfileshouldnotexist" dummy target in program execution error output

WasabiFan, 2022-01-07:
My team just ran into a failure, from which the entirety of the output beyond scons boilerplate was:

```
scons: *** [thisfileshouldnotexist] Error 3221225785
```

Which is... well... _lovely_ 😆 

The "Error 3221225785" bit I/we are still looking into and is probably our fault (the built executable being run appears to be buggy). But _boy_, it's hard to think of a more concerning string to see in an unexplained error than "thisfileshouldnotexist".

I take it, the call below needs to provide _something_ for a "target file" field describing the thing being built and the author opted for a placeholder to act as a phony target. This seems fine but isn't great for clarity in errors.

Is there a better string/configuration here which would mean it still always runs the target but explains better where the error is coming from if it returns nonzero status? This is mostly an scons question, but dependent on what y'all find acceptable for modm.

https://github.com/modm-io/modm/blob/02660d04bf633ae619daa52ce5e19d24e546ede9/tools/build_script_generator/scons/site_tools/utils.py#L66-L69

salkinium, 2022-01-08:
Oh year, this is really old code that we inherited from even before xpcc times.

The easiest fix is to change the string to something useful like: `DummyTarget`.
The harder fix is to find out how to run dummy targets in SCons correctly, we kinda do this for OpenOCD, which you can call over and over again.

Feel free to choose the easy path, SCons documentation is a bit… annoying.

salkinium, 2022-04-13:
This has been fixed in https://github.com/modm-io/modm/commit/98a24832b310056f5b2ba0b58dcfc69cdd38b835

closed 2022-04-13 by salkinium