modm-io/modm #818: STM32F469I-DISCO touch does not work on newer board revisions (B-03)

rleh, 2022-02-04:
ST exchanged the touch screen controller [FT6206](https://modm.io/reference/module/modm-driver-ft6x06/) ([datasheet](https://cdn-shop.adafruit.com/datasheets/FT6x06+Datasheet_V0.1_Preliminary_20120723.pdf)) by FT6336 ([datasheet](https://focuslcds.com/content/FT6236.pdf)).

Touch controller on rev B-01:
[image]
[image]

Touch controller on rev B-03:
[image]
[image]

---

The driver is basically the same, but the I²C address has changed from `0x2a` to `0x38`. Should get fixed in the STM32F469I-DISCO board support package / examples.
I'm not aware of any way to detect the board revision from software, maybe we could try to ping the touch screen controller during initialization and then use the I²C address that responded?
Otherwise we could duplicate the lbuild board module or add a revision option to the lbuild module?

cc @kikass13 @macslow

kikass13, 2022-02-04:
> I'm not aware of any way to detect the board revision from software, maybe we could try to ping the touch screen controller during initialization and then use the I²C address that responded?

yeah, bus scan the shit out of that 7 bit adress space ! :D

salkinium, 2022-02-04:
Each `I2cDevice` has a `setAddress()` function to change the address at runtime. You can `RF_CALL_BLOCKING(ping())` and then change the address inside the right `Board::initialize()` function?

salkinium, 2022-02-04:
If that fails you could always add an option to the BSP module of course that would add the right driver dependency, but it won't be as elegant as adding the `/revN` to the target string.
Maybe something like `<extends>modm:disco-f469/revb03</extends>` would be really cool, I could probably hack that into lbuild, but it could also be achieved by having a second bsp module: `<extends>modm:disco-f469-b03</extends>`.
It would be cool though to always declare the revision by default (which is really part of the board identifier) and then get fixes later for free (just like with the `/revN` target string). Currently you're out of luck, because you *must* use `modm:disco-f469` since `modm:disco-f469-b03` doesn't exist, so lbuild could ignore the `/revb03` part if it doesn't find it.

salkinium, 2022-02-04:
I'm basically just talking myself into implementing this in lbuild. I've already made up my mind, I just need to rationalize it… THE BLUEPILL/REDPILL CHOICE IS AN ILLUSION!1!!

salkinium, 2022-02-06:
> You can `RF_CALL_BLOCKING(ping())` and then change the address inside the right `Board::initialize()` function?

I forgot that the touch sensor is optional and therefore not instantiated by default, therefore the application needs to set the address.
I've added proper config version support and added a `Board::ft6::TouchAddress` that changes depending on the config version that the application must set.

PR #819 (2022-02-14): [board] Add configuration version for DISCO-F469NI
This adds configuration versions for the DISCO-F469NI board `<extends>modm:disco-f469ni:b-03</extends>` to fix the different I2C address of the touch sensor.
See https://github.com/modm-io/modm/issues/818#issuecomment-1030415850.

Fixes #818.

- [x] Depends on https://github.com/modm-io/lbuild/pull/73
- [x] Update documentation
- [x] Test in hardware

cc @rleh
closed 2022-02-14 by salkinium