modm-io/modm #834: OpenOCD not flashing Blue Pill

StevenMacias, 2022-03-02:
Hi, 
I am using Arch Linux and I don't remember modifying my setup too much, but now it is impossible to me to flash a bluepill. 

`openocd --version` outputs `Open On-Chip Debugger 0.11.0`

lbuild and scons seem to work properly. Also the command `st-info --probe` outputs the following output: 

```
Found 1 stlink programmers
  version:    V2J17S4
  serial:     303030303030303030303031
  flash:      65536 (pagesize: 1024)
  sram:       20480
  chipid:     0x0410
  descr:      F1xx Medium-density
```

So there is USB connection and both the micro-controller and the programmer can talk. However when I run `scons program` I only see this:

```
➤ scons program
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
╭────────────── /home/steven/Workspace/git/others/modm/build/blue_pill_f103/rtt/scons-release/rtt.elf
╰───OpenOCD───> stm32f103c8t6
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
scons: done building targets.
```
No indicator LEDs blink in the programmer. Also if I uninstall openocd then naturally complains about openocd not being an existing command. 

```
➤ scons program
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
╭────────────── /home/steven/Workspace/git/others/modm/build/blue_pill_f103/rtt/scons-release/rtt.elf
╰───OpenOCD───> stm32f103c8t6
/bin/sh: line 1: openocd: command not found
scons: done building targets.
```

I have been stuck for a couple of weeks with this. It used to work so that is why I am so confused. Any feedback or advice is more than welcome! Thanks.

rleh, 2022-03-02:
What happens if you try call openocd directly?
```bash
openocd -f interface/stlink.cfg -c "transport select hla_swd" -f target/stm32f1x.cfg
```

StevenMacias, 2022-03-02:
Thanks for the fast reply!

```
➤ openocd -f interface/stlink.cfg -c "transport select hla_swd" -f target/stm32f1x.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
fish: Job 1, 'openocd -f interface/stlink.cfg…' terminated by signal SIGSEGV (Address boundary error)
```
This is the result...

chris-durand, 2022-03-02:
> `fish: Job 1, 'openocd -f interface/stlink.cfg…' terminated by signal SIGSEGV (Address boundary error)`

That is a crash of openocd. Something seems not right with your openocd installation.

Could you try upgrading packages? On Arch it is pretty easy to accidentally install a package without upgrading the required dependencies, `pacman -Sy <package name>` would cause exactly that situation.

If that does not help, you can try compiling the version [from github](https://github.com/openocd-org/openocd) yourself. There is also an `openocd-git` package in AUR.

rleh, 2022-03-02:
> `Job 1, 'openocd -f interface/stlink.cfg…' terminated by signal SIGSEGV (Address boundary error)`

That should not happen, looks like your openocd has some issues.
This is rather a Arch Package bug then a modm bug, you probably should report it there: https://bugs.archlinux.org/?project=5&string=openocd

For now you could try if you problem disappears with this binary distribution of OpenOCD: https://xpack.github.io/openocd/

Actually, the SIGSEGV error should also be displayed when called via the modm wrapper, but probably that gets lost somewhere. Maybe @salkinium has some idea?

salkinium, 2022-03-03:
It gets lost because we start OpenOCD in the background somehow with `os.setsid` so that Ctrl-C gets routed to GDB rather than quitting OpenOCD too. I'm not sure how it works, I adapted this from RIOT OS and it worked. Not sure why or how.
https://github.com/modm-io/modm/blob/develop/tools/modm_tools/openocd.py#L97

StevenMacias, 2022-03-09:
Thank you so much for your feedback. I will follow your suggestions and get back to you.

StevenMacias, 2022-04-02:
I updated my system again it it seems to work flawlessly. I guess it can be closed. Thanks for the support!

closed 2022-04-02 by salkinium