modm-io/modm #836: Deprecated `pathlib` dependency in PyPI `modm`

hshose, 2022-03-13:
The pathlib dependency in PyPI modm doesn't work with Python 3.10 and will not be ported (see [here](https://pypi.org/project/pathlib/)).

Therefore the `pip3 install modm` from the installation instructions fails.

The `pathlib2` module linked as successor linked in the PyPI page above seems to work, when I install with `pip3 install lbuild lxml pyelftools pathlib2` instead of `pip3 install modm`.

So probably just change the dependency to pathlib2?

salkinium, 2022-03-13:
Is pathlib not part of the Python 3 standard library? I think this dependency should be removed.

hshose, 2022-03-13:
That seems to be the case, everything seems to work without pathlib/2 so far

salkinium, 2022-03-14:
I've removed the pathlib dependency with modm v0.1.2. The package source code can now be found at https://github.com/modm-ext/modm-cli instead of just my local machine ;-P

closed 2022-03-14 by salkinium