modm-io/modm #841: I2cMaster::connect<Signals..., reset>() without reseting devices

rleh, 2022-03-22:
The `I2cMaster::connect<Signals..., reset>()` function takes a `I2cMaster::ResetDevices` enum value as (last) template argument to disable the [integrated reset mechanism](https://docs.modm.io/develop/api/nucleo-f429zi/structmodm_1_1I2c.html#a1855002ad6a0992e32db4f3ebf0432da).

https://github.com/modm-io/modm/blob/0992ab10ca04557964566e343567d982b99de13c/src/modm/platform/i2c/stm32/i2c_master.hpp.in#L43-L45

The default value is `ResetDevices::Standard` (= pull SDA low and 9 clock cycles at 100kHz) which works fine, but it is not possible to actually specify a (different) value:

```
error: no matching function for call to 'modm::platform::I2cMaster1::connect<modm::platform::detail::DataB8::Scl, modm::platform::detail::DataB9::Sda, modm::I2cMaster::ResetDevices::NoReset>()'
MyI2cMaster::connect<Scl::Scl, Sda::Sda, modm::I2cMaster::ResetDevices::NoReset>();
```

chris-durand, 2022-03-22:
This does not work in C++. The compiler will interpret all parameters as types which are part of the `Signals` parameter pack:
https://gcc.godbolt.org/z/rY46d5M8E

We should make the reset setting a normal runtime function parameter.

salkinium, 2022-03-23:
> We should make the reset setting a normal runtime function parameter.

Yes. It'll be optimized away if unused.

rleh, 2022-03-23:
I'll take care of a fix (before 2022q1 release next week). 🙂

chris-durand, 2022-03-29:
Will be fixed in #842

closed 2022-04-02 by rleh