modm-io/modm #874: Build issue stemming from std::floor constexpr usage

Nicba1010, 2022-06-22:
Compiler used: `gcc-arm-none-eabi-10.3-2021.10-x86_64-linux`
modm commit hash: `c949daff`
Build tool: `cmake 3.23.2`

The gist of the issue seems to be that `std::floor` has only been constexpr including and after `C++23`. Am I missing something obvious here?

Build trace:
```
[1/1] Building CXX object modm/CMakeFiles/modm.dir/src/modm/board/board.cpp.obj
FAILED: modm/CMakeFiles/modm.dir/src/modm/board/board.cpp.obj 
/home/roberto/gcc-arm-none-eabi-10.3-2021.10-x86_64-linux/gcc-arm-none-eabi-10.3-2021.10/bin/arm-none-eabi-g++ -DMODM_DEBUG_BUILD -DSTM32G0 -isystem /home/roberto/ModmPlayground/app/stm32_w6100/modm/ext -isystem /home/roberto/ModmPlayground/app/stm32_w6100/modm/ext/cmsis/core -isystem /home/roberto/ModmPlayground/app/stm32_w6100/modm/ext/cmsis/device -isystem /home/roberto/ModmPlayground/app/stm32_w6100/modm/src -mthumb -mabi=aapcs -ffreestanding -fno-common -finline-small-functions -findirect-inlining -fstack-protector-strong  -Wpedantic -Wextra -Wshadow=local -Wall -Wdouble-promotion -fno-common -Wconversion -ffunction-sections -fdata-sections -fno-strict-aliasing -fshort-enums -fno-builtin -Wunknown-pragmas -Wswitch-enum --std=c++20 -fconcepts-diagnostics-depth=5 -fno-exceptions -fno-rtti -fno-use-cxa-atexit -Og -g3 -fdiagnostics-color=always -mcpu=cortex-m0plus -mthumb -fdata-sections -ffile-prefix-map=/home/roberto/gcc-arm-none-eabi-10.3-2021.10-x86_64-linux/gcc-arm-none-eabi-10.3-2021.10=. -ffile-prefix-map=/home/roberto/ModmPlayground/app/stm32_w6100=. -ffunction-sections -finline-limit=10000 -fshort-wchar -fsingle-precision-constant -funsigned-bitfields -funsigned-char -fwrapv -g3 -gdwarf-3 -fno-move-loop-invariants -fno-split-wide-types -fno-tree-loop-optimize -fno-exceptions -fno-rtti -fno-unwind-tables -fno-use-cxa-atexit -fstrict-enums -std=c++20 -W -Wall -Wdouble-promotion -Wduplicated-cond -Werror=format -Werror=maybe-uninitialized -Werror=overflow -Werror=sign-compare -Wextra -Wlogical-op -Wno-redundant-decls -Wpointer-arith -Wundef -Wno-volatile -Woverloaded-virtual -MD -MT modm/CMakeFiles/modm.dir/src/modm/board/board.cpp.obj -MF modm/CMakeFiles/modm.dir/src/modm/board/board.cpp.obj.d -o modm/CMakeFiles/modm.dir/src/modm/board/board.cpp.obj -c /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.cpp
In file included from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform.hpp:16,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.hpp:14,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.cpp:12:
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform/clock/systick_timer.hpp: In instantiation of 'static void modm::platform::SysTickTimer::initialize() [with SystemClock = Board::SystemClock; modm::percent_t tolerance = (modm::percent_t)0]':
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.hpp:154:40:   required from here
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform/clock/systick_timer.hpp:53:49: error: 'static constexpr modm::GenericPrescaler<T>::Result modm::GenericPrescaler<T>::from_range(T, T, uint32_t, uint32_t) [with T = long long unsigned int; uint32_t = long unsigned int]' called in a constant expression
   53 |    constexpr auto result = Prescaler::from_range(
      |                            ~~~~~~~~~~~~~~~~~~~~~^
   54 |      SystemClock::Frequency/8, 1000, 1, (1ul << 24)-1);
      |      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform/clock/systick_timer.hpp:18,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform.hpp:16,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.hpp:14,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.cpp:12:
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/math/algorithm/prescaler.hpp:135:2: note: 'static constexpr modm::GenericPrescaler<T>::Result modm::GenericPrescaler<T>::from_range(T, T, uint32_t, uint32_t) [with T = long long unsigned int; uint32_t = long unsigned int]' is not usable as a 'constexpr' function because:
  135 |  from_range(T input_frequency,
      |  ^~~~~~~~~~
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/math/algorithm/prescaler.hpp:140:64: error: call to non-'constexpr' function 'double floor(double)'
  140 |   const uint32_t prescaler_floor = std::max(uint32_t(std::floor(desired)), begin);
      |                                                      ~~~~~~~~~~^~~~~~~~~
In file included from /home/roberto/gcc-arm-none-eabi-10.3-2021.10-x86_64-linux/gcc-arm-none-eabi-10.3-2021.10/arm-none-eabi/include/c++/10.3.1/cmath:45,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform/core/delay_impl.hpp:18,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/architecture/interface/delay.hpp:64,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/architecture/interface.hpp:23,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/architecture.hpp:18,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform.hpp:13,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.hpp:14,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.cpp:12:
/home/roberto/gcc-arm-none-eabi-10.3-2021.10-x86_64-linux/gcc-arm-none-eabi-10.3-2021.10/arm-none-eabi/include/math.h:95:15: note: 'double floor(double)' declared here
   95 | extern double floor (double);
      |               ^~~~~
In file included from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform.hpp:16,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.hpp:14,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.cpp:12:
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform/clock/systick_timer.hpp:55:84: error: no matching function for call to 'modm::PeripheralDriver::assertBaudrateInTolerance<result.modm::GenericPrescaler<long long unsigned int>::Result::frequency, 1000, (modm::percent_t)0>()'
   55 |    PeripheralDriver::assertBaudrateInTolerance< result.frequency, 1000, tolerance >();
      |    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
In file included from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/architecture/interface/uart.hpp:18,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/architecture/interface.hpp:27,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/architecture.hpp:18,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform.hpp:13,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.hpp:14,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.cpp:12:
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/architecture/interface/peripheral.hpp:89:2: note: candidate: 'template<long unsigned int available, long unsigned int requested, modm::percent_t tolerance> static void modm::PeripheralDriver::assertBaudrateInTolerance()'
   89 |  assertBaudrateInTolerance()
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/architecture/interface/peripheral.hpp:89:2: note:   template argument deduction/substitution failed:
In file included from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform/uart/uart_hal_2.hpp:133,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform/uart/uart_2.hpp:23,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform.hpp:32,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.hpp:14,
                 from /home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/board/board.cpp:12:
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/platform/uart/uart_hal_2_impl.hpp: In instantiation of 'static void modm::platform::UsartHal2::initialize(modm::platform::UartBase::Parity, modm::platform::UartBase::WordLength) [with SystemClock = Board::S [... cut ...]
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/architecture/interface/peripheral.hpp:89:2: note: candidate: 'template<long unsigned int available, long unsigned int requested, modm::percent_t tolerance> static void modm::PeripheralDriver::assertBaudrateInTolerance()'
   89 |  assertBaudrateInTolerance()
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/roberto/ModmPlayground/app/stm32_w6100/modm/src/modm/architecture/interface/peripheral.hpp:89:2: note:   template argument deduction/substitution failed:
ninja: build stopped: subcommand failed.
```

chris-durand, 2022-06-22:
Hi @Nicba1010,

gcc provides more `constexpr` math functions as an extension to the standard since it introduced support for C++14. Some of these will now make it into C++23. See [here](https://gcc.godbolt.org/z/Mocxe3Mc8) that using `std::floor` in a `constexpr` context generally compiles with the same compiler you are using.

Did you get your toolchain directly from arm? Have you tried compiling some of our stm32 examples with the default scons build system? There the same code is being built and should work fine. We have seen faulty gcc builds provided by Linux distributions in the past. If the examples work we can at least rule out the toolchain installation.

Is `SystemClock::Frequency` defined as `constexpr` in your code? The input parameters not being a constant expression could be the reason why compile-time evaluation is not possible.

We would probably need more context to debug this further. Could you put together a minimal example to reproduce the error?

Nicba1010, 2022-06-22:
I'm just building the `modm` target right now.
I'll try compile the basic example from godbolt you provided.
I'll also give compiling with scons a try, I went straight to cmake.
The toolchain was downloaded directly from arm.