modm-io/modm #876: spi_dma example not compiling?

hshose, 2022-07-10:
If I try to compile the [spi dma example](https://github.com/modm-io/modm/tree/develop/examples/nucleo_f439zi/spi_dma), I get the following error:

```
$ lbuild build && scons
# [...]
In file included from modm/src/modm/platform/spi/spi_master_1_dma.hpp:97,
                 from modm/src/modm/platform.hpp:38:
modm/src/modm/platform/spi/spi_master_1_dma_impl.hpp: In instantiation of 'static modm::ResumableResult<void> modm::platform::SpiMaster1_Dma<DmaChannelRx, DmaChannelTx>::transfer(const uint8_t*, uint8_t*, std::size_t) [with DmaChannelRx = modm::platform::DmaController<2>::Channel<modm::platform::DmaBase::Channel::Stream0>; DmaChannelTx = modm::platform::DmaController<2>::Channel<modm::platform::DmaBase::Channel::Stream3>; uint8_t = unsigned char; std::size_t = unsigned int]':
main.cpp:47:4:   required from here
modm/src/modm/platform/spi/spi_master_1_dma_impl.hpp:99:16: error: 'uint8_t modm::platform::SpiMaster1::state' is private within this context
   99 |         switch(state & Bit1)
      |                ^~~~~
In file included from modm/src/modm/platform.hpp:37:
modm/src/modm/platform/spi/spi_master_1.hpp:41:24: note: declared private here
   41 |         static uint8_t state;
      |                        ^~~~~
modm/src/modm/platform/spi/spi_master_1_dma_impl.hpp:99:16: error: 'uint8_t modm::platform::SpiMaster1::state' is private within this context
   99 |         switch(state & Bit1)
      |                ^~~~~
modm/src/modm/platform/spi/spi_master_1.hpp:41:24: note: declared private here
   41 |         static uint8_t state;
      |                        ^~~~~
modm/src/modm/platform/spi/spi_master_1_dma_impl.hpp:103:17: error: 'uint8_t modm::platform::SpiMaster1::state' is private within this context
  103 |                 state |= Bit1;
      |                 ^~~~~
modm/src/modm/platform/spi/spi_master_1.hpp:41:24: note: declared private here
   41 |         static uint8_t state;
      |                        ^~~~~
modm/src/modm/platform/spi/spi_master_1_dma_impl.hpp:103:17: error: 'uint8_t modm::platform::SpiMaster1::state' is private within this context
  103 |                 state |= Bit1;
      |                 ^~~~~
modm/src/modm/platform/spi/spi_master_1.hpp:41:24: note: declared private here
   41 |         static uint8_t state;
      |                        ^~~~~
modm/src/modm/platform/spi/spi_master_1_dma_impl.hpp:147:17: error: 'uint8_t modm::platform::SpiMaster1::state' is private within this context
  147 |                 state &= ~Bit1;
      |                 ^~~~~
modm/src/modm/platform/spi/spi_master_1.hpp:41:24: note: declared private here
   41 |         static uint8_t state;
      |                        ^~~~~
modm/src/modm/platform/spi/spi_master_1_dma_impl.hpp:147:17: error: 'uint8_t modm::platform::SpiMaster1::state' is private within this context
  147 |                 state &= ~Bit1;
      |                 ^~~~~
modm/src/modm/platform/spi/spi_master_1.hpp:41:24: note: declared private here
   41 |         static uint8_t state;
      |                        ^~~~~
scons: *** [/home/user/projects/modm_projects/modm/build/nucleo_f439zi/spi_dma/scons-release/main.o] Error 1
scons: building terminated because of errors.
```
Build with recent develop branch and arm-none-eabi-gcc 12.1.0-1 on arch.
Am I missing something? CI seems to build the example...

rleh, 2022-07-10:
I can not reproduce the error with ARMs official GCC (version 10.2.1) on my Fedora machine.

Could you please try if the latest ARM GCC from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads works?

chris-durand, 2022-07-10:
@rleh I can reproduce this with the 2022.02 GCC 11.2 ARM toolchain but the 10.2 works for me.

```
modm/src/modm/platform/spi/spi_master_1_dma_impl.hpp: In instantiation of 'static modm::ResumableResult<void> modm::platform::SpiMaster1_Dma<DmaChannelRx, DmaChannelTx>::transfer(const uint8_t*, uint8_t*, std::size_t) [with DmaChannelRx = modm::platform::DmaController<2>::Channel<modm::platform::DmaBase::Channel::Stream0>; DmaChannelTx = modm::platform::DmaController<2>::Channel<modm::platform::DmaBase::Channel::Stream3>; uint8_t = unsigned char; std::size_t = unsigned int]':
main.cpp:47:4:   required from here
modm/src/modm/platform/spi/spi_master_1_dma_impl.hpp:99:16: error: 'uint8_t modm::platform::SpiMaster1::state' is private within this context
   99 |         switch(state & Bit1)
      |                ^~~~~
```

rleh, 2022-07-10:
With ARMs GCC 11.2 I get another error:
```
$ cd examples/nucleo_f439zi/spi_dma
& lbuild build
$ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Compiling C++·· release/main.o
In file included from /opt/user/gcc-arm-11.2-2022.02-x86_64-arm-none-eabi/arm-none-eabi/include/c++/11.2.1/chrono:41,
                 from modm/src/modm/architecture/interface/clock.hpp:21,
                 from modm/src/modm/architecture/interface.hpp:22,
                 from modm/src/modm/architecture.hpp:18,
                 from modm/src/modm/platform.hpp:13,
                 from modm/src/modm/board/board.hpp:15,
                 from modm/src/modm/board.hpp:12,
                 from main.cpp:12:
/opt/user/gcc-arm-11.2-2022.02-x86_64-arm-none-eabi/arm-none-eabi/include/c++/11.2.1/limits:1673:7: internal compiler error: Illegal instruction
 1673 |       min() _GLIBCXX_USE_NOEXCEPT { return __FLT_MIN__; }
      |       ^~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
scons: *** [/home/user/modm/build/nucleo_f439zi/spi_dma/scons-release/main.o] Error 1
scons: building terminated because of errors.
```

hshose, 2022-07-10:
@chris-durand just tried with GCC gcc-arm-11.2-2022.02-x86_64-arm-none-eabi binary from arm website and I get the same error... I'll try the more historic gcc versions...

chris-durand, 2022-07-10:
@hshose The GCC 10.2 2021.10 version from ARM will work. That is what I am using with modm at work and what the CI currently compiles with.

chris-durand, 2022-07-10:
The compiler is completely right rejecting that code. The variable `state` is defined as `private` in the base class `SpiMaster1`. You should not be able to access it from a derived class:

```
class SpiMaster1 : public modm::SpiMaster
{
	static uint8_t state;
```

GCC had a bug that caused that code to compile previously: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=58993

chris-durand, 2022-07-10:
I'll make a PR in a few minutes to fix it.

chris-durand, 2022-07-10:
@rleh @hshose https://github.com/modm-io/modm/pull/877

PR #877 (2022-07-13): [stm32] Fix compilation of SPI DMA driver with GCC 11+
Fix incorrect access specifier in `SpiMaster` that prevents `SpiMasterDma` from compiling with GCC 11.

GCC versions prior to 11 allowed accessing private member variables of template base classes from sub-classes: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=58993

cc @hshose 

Fixes #876
closed 2022-07-13 by salkinium