modm-io/modm #888: BluePill PWM Generation Example Code Not Working

StevenMacias, 2022-07-23:
Hi! 
I am using the code in `modm/examples/nucleo_g474re/servo_pwm/main.cpp` as a inspiration for a PWM code for the BluePill.
I have used a ProtoThread and it looks like this: 

```cpp
#define PWM_CHANNEL 2
#define PWM_FREQ_HZ 50
#define PWM_MS_PER_PERIOD (1000/PWM_FREQ_HZ)

#include "PwmThread.hpp"
#include <modm/board.hpp>

constexpr uint16_t overflow = 0xffff;
constexpr uint16_t minPwm = (overflow / PWM_MS_PER_PERIOD) * 1.f; // 1 ms pulse width
constexpr uint16_t targetPwm = (overflow / PWM_MS_PER_PERIOD) * 1.5f; // 1.5 ms pulse width
constexpr uint16_t maxPwm = (overflow / PWM_MS_PER_PERIOD) * 2.f; // 2 ms pulse width

bool PwmThread::run()
{
    PT_BEGIN();
        MODM_LOG_INFO << "PWM_FREQ_HZ: " << PWM_FREQ_HZ << modm::endl;
        MODM_LOG_INFO << "PWM_MS_PER_PERIOD: " << PWM_MS_PER_PERIOD << modm::endl;
        
        // Code used to test that we are in the correct GPIO
        if(test_gpio == true)
        {
            // TEST GPIO PIN
            GpioB7::setOutput();
            GpioB7::set();
        }
        // Code used to initialize PWM
        else
        {
            // TEST PWM
            pwmInit();
        }
        
        while (true)
        {
            if(test_gpio == false)
            {
                // TEST PWM
                Timer4::setCompareValue(PWM_CHANNEL, targetPwm);
            }
            PT_YIELD();
        }
    PT_END();
}

void PwmThread::pwmInit()
{
    Timer4::connect<GpioB7::Ch2>();
    Timer4::enable();
    Timer4::setMode(Timer4::Mode::UpCounter);
    
    // 170 MHz / 52 / 2^16 = approx. 50 Hz -> NUCLEO G474RE
    // 72Mhz  / X  / 2^16 = approx. 50 Hz -> Blue Pill
    // X = 72MHz/(50Hz*2^16) = 21

    Timer4::setPrescaler(21);
    Timer4::setOverflow(overflow);

    Timer4::configureOutputChannel(PWM_CHANNEL, Timer4::OutputCompareMode::Pwm, 0);

    Timer4::applyAndReset();
    Timer4::start();
    Timer4::enableOutput();
}

uint16_t PwmThread::getValue()
{
    return Timer4::getValue();
}

```
In the oscilloscope I just get a flat line. I even started doubting about my own existence so that is why you can find the private variable `test_gpio` to set the GPIO to High and make sure that the oscilloscope is hooked to the right pin. 

I adapted the prescaler value so it generates a 50Hz frequency in the BluePill too. However, this is the result I get: 
[image]

I managed to get my hands on a Nucleo-G474RE and with minimal changes (commenting the RTT lines, changing the LED name and adjusting the project.xml accordingly) I managed to get a PWM with the same code: 
[image]
Note that the frequency is 123Hz since I used the values for the 72Mhz clock of the BluePill. 

I call the `getValue()` function from an menu thread that I am able to access through RTT, these are the logs that prove that the BluePill Timer is actually counting: 
```
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
╭───OpenOCD───> Real Time Transfer
╰─────RTT────── stm32f103c8t6
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J17S4 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.268692
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : rtt: Searching for control block 'modm.rtt.modm'
Info : rtt: Control block found at 0x20000c00
Info : Listening on port 9090 for rtt connections

---- MENU ----
 0: Toggle LED activity
 +: Increase LED pulse time
 -: Decrease LED pulse time
 d: set default LED pulse time
 t: show Timer4 value
 m: Print this menu 

PWM_FREQ_HZ: 50
PWM_MS_PER_PERIOD: 20
t
Timer4 value: 47181
t
Timer4 value: 27213
ttt
Timer4 value: 9575
Timer4 value: 9680
Timer4 value: 9785
t
Timer4 value: 13666
```
I checked the Datasheet and the Reference Manual for the STM32F103 but I was not able to find an obvious answer for my mistake.

Any idea what I am doing wrong? 

Thanks!

salkinium, 2022-07-23:
cc @rleh @chris-durand have more experience in this area of the HAL.

chris-durand, 2022-07-23:
It was totally non-obvious for me at first but I think I have found it after digging through the reference manual and the modm GPIO source code. My first suspicion was right that this has to do with the odd F1 pin remapping design.

On all STM32 devices except F1 there is only one alternate function mode for both inputs and outputs. The pin direction is controlled by the peripheral.

On STM32F1 you have to manually select whether your alternate function pin is an input or output:
[image]

For input alternate functions standard input mode has to be selected.

In case of the timer the modm API has no way to tell which one you want by writing `Timer4::connect<GpioB7::Ch2>();` since the timer pin can be either an input or output.

The modm API will only set the pin to alternate function output in case it already has been configured as output before:
```
inline static void setAlternateFunction()
{
    uint32_t cr = *gcr();
    if (cr & (0b0011UL << (cr_pin * 4)))
        *gcr() = cr | (0b1000UL << (cr_pin * 4));
}
```
(The code refers to the bits from the reference manual shown above).

If I am right putting `GpioB7::setOutput()` before `Timer4::connect<GpioB7::Ch2>()` will fix the problem.

Another thing seems odd. The modm API has `Timer4::enableOutput()` which sets the `BDTR` register but that should not exist for F1 general purpose timers. The device header uses the same register definitions for all types of timers even if they lack certain functions. I don't think it causes any issues here but we could fix that in the modm driver nevertheless.

StevenMacias, 2022-07-24:
Wow, It worked. Thank you so much!
`GpioB7::setOutput();` before `Timer4::connect<GpioB7::Ch2>();` did the trick.
[image]
Does it makes sense to add the `setOutput()` inside the `connect()` for the code generated for the STM32F1? 
If so, what steps should I follow to contribute with that modification? Never worked with the modm HAL before. 
Thanks again!

salkinium, 2022-07-24:
> Does it makes sense to add the `setOutput()` inside the `connect()` for the code generated for the STM32F1?

The problem is that you can also use the timer channel in input capture mode and thus requires input.
Something like `Timer4::connect<GpioB7::Ch2Out>();` and `Timer4::connect<GpioB7::Ch2In>();` may be required, but then  you're intentionally mixing two APIs that are technically completely separate.

I think the solution is to document the Timer API properly, because there are a number of other pitfalls like forgetting to `enableOutput()`.

StevenMacias, 2022-07-24:
ok! then for the time being I could just prepare a PWM modm example for the BluePill and adding a comment linking to this issue in the `GpioB7::setOutput();` line?