modm-io/modm #906: STM32 Timer: connect and configureOutputChannel

strongly-typed, 2022-09-02:
`connect` uses a symbolic template parameter to connect to the pin:

[`Timer4::connect<GpioB7::Ch2>();`](https://github.com/modm-io/modm/blob/develop/examples/blue_pill_f103/servo_pwm/main.cpp#L40)

whereas `configureOutputChannel` uses a plain int for this.

[`Timer4::configureOutputChannel(2, Timer4::OutputCompareMode::Pwm, 0);`](https://github.com/modm-io/modm/blob/develop/examples/blue_pill_f103/servo_pwm/main.cpp#L40)

I would love to create a template class that gets `GpioB7::Ch2` as a template parameter and then uses this for `connect` (possible) *and* deduced the channel number `2` from the template parameter `GpioB7::Ch2`.

Is there any chance to get something like `GpioB7::Ch2::ChId` or similar that can be passed to `configureOutputChannel`?

chris-durand, 2022-09-02:
I have made a proof-of-concept on [that branch](https://github.com/chris-durand/modm/tree/feature/stm32_timer_channel_signal).

With that change you can now also write:
```
Timer4::configureOutputChannel<GpioB7::Ch2>(Timer4::OutputCompareMode::Pwm, 0);
Timer4::setCompareValue<GpioB7::Ch2>(1000);
```
This only implements it for those two methods in the general-purpose timer implementation but feel free to take my changes and extend them if you like.

salkinium, 2022-09-03:
Nice, though it would have to be for backward compatibility:
```cpp
Timer4::configureOutputSignal<GpioB7::Ch2>(Timer4::OutputCompareMode::Pwm, 0);
Timer4::setCompareValueSignal<GpioB7::Ch2>(1000);
```
or alternatively you could also do this less elegant version:
```cpp
Timer4::configureOutputChannel(Timer4::signalToChannel<GpioB7::Ch2>(), Timer4::OutputCompareMode::Pwm, 0);
```