modm-io/modm #923: Does not seem to work with gcc 12.2.0

linasdev, 2022-11-01:
Freshly cloned the repository, then:
```
$ git submodule update --init
$ cd examples/blue_pill_f103/blink
$ lbuild build
$ scons
```

Environment:
```
$ uname -a
Linux linas-pc 6.0.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 29 Oct 2022 14:08:39 +0000 x86_64 GNU/Linux
```
```
$ arm-none-eabi-ld --version
GNU ld (GNU Binutils) 2.39
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Build output:

```
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Compiling C++·· release/main.o
Compiling C···· release/modm/ext/gcc/cabi.o
Compiling C++·· release/modm/ext/gcc/cxxabi.o
Compiling C++·· release/modm/ext/gcc/new_delete.o
Compiling C++·· release/modm/src/modm/architecture/driver/atomic/flag.o
Compiling C++·· release/modm/src/modm/board/board.o
Compiling C++·· release/modm/src/modm/math/utils/bit_operation.o
Compiling C++·· release/modm/src/modm/math/utils/pc/operator.o
Compiling C++·· release/modm/src/modm/platform/clock/rcc.o
Compiling C++·· release/modm/src/modm/platform/clock/systick_timer.o
Compiling C++·· release/modm/src/modm/platform/core/assert.o
Compiling C++·· release/modm/src/modm/platform/core/delay.o
Compiling C++·· release/modm/src/modm/platform/core/delay_ns.o
Compiling C···· release/modm/src/modm/platform/core/no_heap.o
Assembling····· release/modm/src/modm/platform/core/reset_handler.o
Compiling C···· release/modm/src/modm/platform/core/startup.o
Compiling C···· release/modm/src/modm/platform/core/startup_platform.o
Compiling C···· release/modm/src/modm/platform/core/vectors.o
Compiling C++·· release/modm/src/modm/platform/gpio/enable.o
Archiving······ release/modm/libmodm.a
Indexing······· release/modm/libmodm.a
Linking········ /home/linasdev/Projects/Embedded/asd/modm/build/blue_pill_f103/blink/scons-release/blink.elf
/usr/lib/gcc/arm-none-eabi/12.2.0/../../../../arm-none-eabi/bin/ld: warning: /home/linasdev/Projects/Embedded/asd/modm/build/blue_pill_f103/blink/scons-release/blink.elf has a LOAD segment with RWX permissions
collect2: error: ld returned 1 exit status
scons: *** [/home/linasdev/Projects/Embedded/asd/modm/build/blue_pill_f103/blink/scons-release/blink.elf] Error 1
scons: building terminated because of errors.
```

linasdev, 2022-11-01:
Works with:
```
$ arm-none-eabi-ld --version
GNU ld (GNU Binutils) 2.28
Copyright (C) 2017 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

chris-durand, 2022-11-02:
Hi @linasdev,

Thanks for reporting the issue. I could reproduce it with the latest gcc 12 toolchain from ARM. The newly added warning in gcc 12 causes linking to fail since we pass `--fatal-warnings` to the linker. It complains about the flash section in the elf file being both executable and writable and emits a warning, I would assume for security reasons. In our case that is kind of pointless since the flash is not writable from a hardware point of view by performing a simple memory access.

The root cause for the message in our case seems to be that we are [placing non-const variables](https://github.com/modm-io/modm/blob/2ba98f59587551275d0bf78a663f1d358450ae48/src/modm/platform/core/cortex/hardware_init.hpp#L22) in a section located in flash.

Could you please check if #924 solves the problem for you?

closed 2022-11-06 by chris-durand

orangesalad, 2023-02-18:
I tried to build an example with latest arm toolchain (12.2.Rel1) and the latest develop branch but sill seen this issue. I downgraded to 11.3.Rel1 and it resolved the issue. @chris-durand not sure if something is still missing from #924

salkinium, 2023-02-18:
It's getting fixed with #940. There's a missing link flag that's not backward compatible with previous binutils (really annoying 🙄)