modm-io/modm #932: Stm32G4: Timer2-5 bug

ser-plu, 2022-12-01:
According to reference manual, Timers 2..5 of Stm32G4 family do not have BDTR register. modm generates functions using this register.

salkinium, 2022-12-02:
Maybe @rleh or @chris-durand know more?

chris-durand, 2022-12-02:
Thanks @ser-plu for pointing out the issue. Of the STM32 general purpose timers only 15, 16 and 17 support complementary PWM with dead-time generation. All the other instances do not have the `BDTR` register. I will open a PR to stop generating those functions for timer instances that do not support them.

@salkinium That information should actually be inside the modm-devices data but we don't support per-instance feature information. Adding a new driver type for every combination of timer features also seems a bit messy. Do you see any clean way of addressing this issue in the future? I will fix it in the driver for now.

closed 2022-12-03 by chris-durand