modm-io/modm #959: macOSarm-gcc-bin@12 from osx-cross failes

strongly-typed, 2023-02-13:
Works:

```
➜  blink git:(develop) ✗ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/local/Cellar/arm-gcc-bin@11/11.3.Rel1/bin/../libexec/gcc/arm-none-eabi/11.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: /Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/src/gcc/configure --target=arm-none-eabi --prefix=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/install --with-gmp=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/host-tools --with-mpfr=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/host-tools --with-mpc=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/host-tools --with-isl=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/host-tools --disable-shared --disable-nls --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-newlib --with-gnu-as --with-gnu-ld --with-sysroot=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/install/arm-none-eabi --with-multilib-list=aprofile,rmprofile --with-pkgversion='Arm GNU Toolchain 11.3.Rel1' --with-bugurl=https://bugs.linaro.org/
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 11.3.1 20220712 (Arm GNU Toolchain 11.3.Rel1)
➜  blink git:(develop) ✗ lbuild build && scons

scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Memory usage··· /Users/sascha/Dev/modm/modm-str/build/stm32f4_discovery/blink/scons-release/blink.elf
Program:   3.1 KiB /   1.0 MiB (0.3% used)
(.fastdata + .rodata + .text)

Data:      3.1 KiB / 196.0 KiB (1.6% used) = 144 B static (0.1%) + 3072 B stack (1.5%)
(.bss + .fastdata + .stack)

Heap:    192.9 KiB / 196.0 KiB (98.4% available)
(.heap_backup + .heap_ccm + .heap_sram1 + .heap_sram2)

scons: done building targets.
```

Does not work:
```
➜  blink git:(develop) ✗ brew unlink arm-gcc-bin@11
Unlinking /usr/local/Cellar/arm-gcc-bin@11/11.3.Rel1... 96 symlinks removed.
➜  blink git:(develop) ✗ brew link arm-gcc-bin@12
Linking /usr/local/Cellar/arm-gcc-bin@12/12.2.Rel1... 96 symlinks created.
➜  blink git:(develop) ✗ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/local/Cellar/arm-gcc-bin@12/12.2.Rel1/bin/../libexec/gcc/arm-none-eabi/12.2.1/lto-wrapper
Target: arm-none-eabi
Configured with: /Volumes/data/jenkins/workspace/GNU-toolchain/arm-12/src/gcc/configure --target=arm-none-eabi --prefix=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-12/build-arm-none-eabi/install --with-gmp=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-12/build-arm-none-eabi/host-tools --with-mpfr=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-12/build-arm-none-eabi/host-tools --with-mpc=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-12/build-arm-none-eabi/host-tools --with-isl=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-12/build-arm-none-eabi/host-tools --disable-shared --disable-nls --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-newlib --with-gnu-as --with-gnu-ld --with-sysroot=/Volumes/data/jenkins/workspace/GNU-toolchain/arm-12/build-arm-none-eabi/install/arm-none-eabi --with-multilib-list=aprofile,rmprofile --with-pkgversion='Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)' --with-bugurl=https://bugs.linaro.org/
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.2.1 20221205 (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24))
➜  blink git:(develop) ✗ lbuild build && scons

scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Compiling C++·· release/main.o
Compiling C···· release/modm/ext/gcc/cabi.o
Compiling C++·· release/modm/ext/gcc/cxxabi.o
Compiling C++·· release/modm/ext/gcc/new_delete.o
Compiling C++·· release/modm/src/modm/architecture/driver/atomic/flag.o
Compiling C++·· release/modm/src/modm/board/board.o
Compiling C++·· release/modm/src/modm/container/smart_pointer.o
Compiling C++·· release/modm/src/modm/io/iostream.o
Compiling C++·· release/modm/src/modm/io/iostream_printf.o
Compiling C++·· release/modm/src/modm/math/utils/bit_operation.o
Compiling C++·· release/modm/src/modm/math/utils/pc/operator.o
Compiling C++·· release/modm/src/modm/platform/clock/rcc.o
Compiling C++·· release/modm/src/modm/platform/clock/systick_timer.o
Compiling C++·· release/modm/src/modm/platform/core/assert.o
Compiling C++·· release/modm/src/modm/platform/core/delay.o
Compiling C++·· release/modm/src/modm/platform/core/delay_ns.o
Compiling C···· release/modm/src/modm/platform/core/no_heap.o
Assembling····· release/modm/src/modm/platform/core/reset_handler.o
Compiling C···· release/modm/src/modm/platform/core/startup.o
Compiling C···· release/modm/src/modm/platform/core/startup_platform.o
[... 63 lines cut ...]
Linking········ /Users/sascha/Dev/modm/modm-str/build/stm32f4_discovery/blink/scons-release/blink.elf
/usr/local/Cellar/arm-gcc-bin@12/12.2.Rel1/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: warning: /Users/sascha/Dev/modm/modm-str/build/stm32f4_discovery/blink/scons-release/blink.elf has a LOAD segment with RWX permissions
collect2: error: ld returned 1 exit status
scons: *** [/Users/sascha/Dev/modm/modm-str/build/stm32f4_discovery/blink/scons-release/blink.elf] Error 1
scons: building terminated because of errors.
```

I am using the toolchain from osx-cross

```
➜  blink git:(develop) ✗ brew info arm-gcc-bin@11
==> osx-cross/arm/arm-gcc-bin@11: stable 11.3.Rel1 (bottled)
Pre-built GNU toolchain for Arm Cortex-M and Cortex-R processors
https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads
/usr/local/Cellar/arm-gcc-bin@11/11.3.Rel1 (6,336 files, 2.9GB)
  Poured from bottle on 2023-02-13 at 22:44:51
From: https://github.com/osx-cross/homebrew-arm/blob/HEAD/Formula/arm-gcc-bin@11.rb
➜  blink git:(develop) ✗ brew info arm-gcc-bin@12
==> osx-cross/arm/arm-gcc-bin@12: stable 12.2.Rel1 (bottled)
Pre-built GNU toolchain for Arm Cortex-M and Cortex-R processors
https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads
/usr/local/Cellar/arm-gcc-bin@12/12.2.Rel1 (6,352 files, 840.7MB) *
  Poured from bottle on 2023-02-13 at 22:34:32
From: https://github.com/osx-cross/homebrew-arm/blob/HEAD/Formula/arm-gcc-bin@12.rb
```

chris-durand, 2023-02-13:
It's a known issue. Linking with GCC12 emits a linker warning that is treated as error. When #940 is merged the warning will be disabled and GCC 12 will work.

strongly-typed, 2023-02-13:
Awwhh, great! Looked in the issues, but not in the PRs.

salkinium, 2023-02-16:
@strongly-typed If you have time and motivation, I would love if you could locally debug the new Docker container for the documentation generator. [Specifically this step](https://github.com/modm-io/modm/blob/develop/.github/workflows/linux.yml#L392) simply never makes/reports progress. I suspect it's something to do with the [file deduplication via symlinks](https://github.com/modm-io/modm/blob/develop/tools/scripts/docs_modm_io_generator.py#L204-L229) in combination with [Pythons multiprocessing issues](https://pythonspeed.com/articles/python-multiprocessing/) (which I've already tried to fix by [switching to `spawn` mode](https://github.com/modm-io/modm/pull/940/commits/5c574c32fb08adf1b0efb8a19c5e9e6fcdc955df)). I cannot reproduce this on my mac, it must be something with the Docker container specifically. Perhaps the upgrade from Ubuntu 20.04 to 22.04 changed something with the filesystem? That's the very last thing that's holding back GCC12 support.

closed 2023-02-18 by salkinium