modm-io/modm #966: STM32F103 PB4 not accessible / remapping not part of modm

strongly-typed, 2023-03-05:
Hi there,

ST decided to use `PB4` as `D5` on their [F103RB nucleo](https://github.com/modm-io/modm/blob/develop/src/modm/board/nucleo64_arduino.hpp#L29).

Unfortunately, `PB4` is by default mapped as `NJTRST`, see [Table 37 of RM0008](https://www.st.com/resource/en/reference_manual/cd00171190-stm32f101xx-stm32f102xx-stm32f103xx-stm32f105xx-and-stm32f107xx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf).

I have not found a way to remap PB4 to GPIO by modm.

In an example with [Blue Pill there is a hint in the code](https://github.com/modm-io/modm/blob/develop/examples/blue_pill_f103/encoder_input/main.cpp#L55) to access `AFIO->MAPR` directly, which works.

What should we do about it? Add new API to modm?

salkinium, 2023-03-05:
Probably not as this is very specialized and there's no added value over just setting the right bit in the register.
I would set the bit in the board initialize function, since there is no need to connect under reset for this board anyways.

salkinium, 2023-03-06:
I mean, yeah, it would be nice to have an API for all sorts of things ;-P
It's just not trivial to implement:

1. We have no immediate data source of pin functions to bit mappings. It's probably different on a lot of devices.
2. Semantically, we'd have to use the `GpioB4::disconnect()` function, but it would disconnect a group of pins, so it has side effects not modelled by the API.
3. Someone is going to disconnect the wrong pin and switch off SWD and then they're locked out of their chip (unless of course they connect under reset, unless they disconnected that too).

Also this is the first time in 10y that I heard about this problem. Connecting the USB pins also requires more configuration than the GPIO API can handle, and it was way easier to just set the registers in the board function instead of thinking about how to properly model this. I'm pretty sure nobody at ST actually thinks about how to write consistent software APIs for their hardware configurations, so I'm hesitant to put work into it, only to have it break later on, because ST added yet another configuration method. \</rant\>

PR #967 (2023-03-06): [board] [stm32] Nucleo F103RB enable PB4
Resolves #966
closed 2023-03-06 by salkinium