modm-io/modm #971: Debugger (arm-none-eabi-gdb) not working on up-to-date linux systems

rleh, 2023-03-12:
Our [installation guide](https://modm.io/guide/installation/) suggests to install the ["official" precompiled GNU toolchain from ARM](https://developer.arm.com/Tools%20and%20Software/GNU%20Toolchain).
If you follow the instructions you will not have a working GDB debugger on modern Linux systems (e.g. Ubuntu ≥22.04 or Fedora ≥36) because ARM apparently requires Python3.8 (which is end-of-life since May 2021).
Also in our docker image(s) `arm-none-eabi-gdb` does not work for the same reason:

```
> docker run --rm -it ghcr.io/modm-ext/modm-build-cortex-m:latest
root@d8fbff209b23:/work# arm-none-eabi-gdb
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = '/usr/local/bld-tools/bld-tools-virtual-env/bin/python'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/usr/local/bld-tools/bld-tools-virtual-env/bin/python'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.executable = '/usr/local/bld-tools/bld-tools-virtual-env/bin/python'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/usr/lib/python38.zip',
    '/usr/lib/python3.8',
    '/usr/lib/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007f9e995ffc00 (most recent call first):
<no Python frame>
```
(Setting `PYTHONHOME`/`PYTHONPATH` does not help because python3.8 is compiled into the `arm-none-eabi-gdb` binary.)

salkinium, 2023-03-12:
Ugh. Additionally I wanted to use `arm-none-eabi-gdb-py`, but they removed that option for GCC11/12 because GCC9/10 still used Python 2.7, which is also EOL (since Jan 2020 even). So instead of upgrading to Python 3.x, they just removed it.

I've been using the xPack gcc at work (on macOS), which is compiled from the official source tarball and comes with Python 3.11. Can you check if that works on Linux too?
https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases

rleh, 2023-03-12:
For all Fedora users (including myself) I created a arm-none-eabi-gdb package that can by easily installed via the package manager: https://copr.fedorainfracloud.org/coprs/rleh/arm-none-eabi-gdb/ (The copr builds are currenlty enabled for Fedora 36, 37, 38 and Rawhide on x86_64 and Aarch64 architectures.)

```
dnf install 'dnf-command(copr)'
dnf copr enable rleh/arm-none-eabi-gdb
dnf install arm-none-eabi-gdb
```

Works perfect with `scons debug` and with the Vscode debugger.

rleh, 2023-03-12:
> Can you check if that works on Linux too?
> https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases

Let's see if it works in our docker container: modm-ext/docker-modm-build/pull/28

rleh, 2023-03-12:
On my Fedora 36 system the xPack toolchain seems to work fine!

rleh, 2023-03-12:
The [xPack site](https://xpack.github.io/blog/2023/02/05/arm-none-eabi-gcc-v12-2-1-1-2-released/#python) states:

> The Python 3 run-time is included, so GDB does not need any version of Python to be installed, and is insensitive to the presence of other versions.

Nice!

salkinium, 2023-03-12:
Yup, and I was thinking of adding a few Python scripts to our GDB setup for protothreads/fiber support, since that is definitely not supported by GDB or OpenOCD, so having an integrated Python library will give us way more control.

I really hope it's a drop-in replacement, we should do some Dissassmbly diffing to see what changed.

PR #972 (2023-04-10): xPack GNU Arm toolchain
- [x] [docs] Installation guide: Use xPack GNU Arm toolchain instead Arm precompiled
  - [x] Linux
  - [ ] macOS
  - [x] Windows
- [x] [ci] Update docker images to `:2023-03-12` version
- [x] Add xPack formula to [modm-ext/homebrew-modm](https://github.com/modm-ext/homebrew-modm): modm-ext/homebrew-modm/pull/1
- [x] Diff the disassembly of a few interesting example
- [x] Test new compiler in hardware
  - [x] Tested by @rleh with STM32U575 and STM32F429
- [x] Don't strip extra compiler version info from [`MODM_BUILD_COMPILER`](https://modm.io/reference/module/modm-build/ [... cut]
closed 2023-04-10 by salkinium