modm-io/modm #984: Problem with SPI Mode 3 on STM32F469

hshose, 2023-03-22:
I'm trying to use SPI in Mode 3 on STM32F469, like this:

<details>

<summary> 

Details (click to see complete `main.cpp`)

```
Cs::reset();
RF_CALL_BLOCKING(SpiMaster::transfer(0xaa));
RF_CALL_BLOCKING(SpiMaster::transfer(nullptr, buffer.data(), 1));
Cs::set();

modm::delay(20us);
SpiMaster::setDataMode(SpiMaster::DataMode::Mode3);
modm::delay(20us);

Cs::reset();
RF_CALL_BLOCKING(SpiMaster::transfer(0xaa));
RF_CALL_BLOCKING(SpiMaster::transfer(nullptr, buffer.data(), 1));
Cs::set();
```

</summary>

```cpp
#include <modm/board.hpp>
#include <modm/processing.hpp>

using namespace Board;

using SpiMaster = modm::platform::SpiMaster2;

using Cs   = D9;
using Mosi = D11;
using Miso = D12;
using Sck  = D13;

int
main()
{
    Board::initialize();
    Cs::setOutput(modm::Gpio::High);

    SpiMaster::connect<Miso::Miso, Mosi::Mosi, Sck::Sck>();
    SpiMaster::initialize<Board::SystemClock, 351_kHz>();

    MODM_LOG_INFO       << "==========SPI Mode 3 The Fuck Test==========" << modm::endl;

    std::array<uint8_t,2> buffer;

    Cs::reset();
    RF_CALL_BLOCKING(SpiMaster::transfer(0xaa));
    RF_CALL_BLOCKING(SpiMaster::transfer(nullptr, buffer.data(), 1));
    Cs::set();

    modm::delay(20us);
    SpiMaster::setDataMode(SpiMaster::DataMode::Mode3);
    modm::delay(20us);

    Cs::reset();
    RF_CALL_BLOCKING(SpiMaster::transfer(0xaa));
    RF_CALL_BLOCKING(SpiMaster::transfer(nullptr, buffer.data(), 1));
    Cs::set();

    while (true)
    {
    
        Board::LedOrange::toggle();
    }

    return 0;
}
```

</details>

In the default (mode 0) everything works fine, however, after I change to mode 3, the CS goes high one clock cycle early and there are another 7 (wtf?) clock cycles.

Im a bit puzzled...

[image]

According to [RM0386](https://www.st.com/resource/en/reference_manual/rm0386-stm32f469xx-and-stm32f479xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), it looks like the `setDataMode` function sets the correct bits in the `SPI_CR1` register...

rleh, 2023-03-22:
I can't reproduce the bug on Nucleo-F429ZI:

[image]

chris-durand, 2023-03-22:
I can reproduce it on a Nucleo F446ZE:
[image]

(The scope decoding function is set up for SPI mode 3. Thus, decoding the first two bytes fails although the data is fine).

The issue is that reconfiguring the SPI mode is not allowed while the peripheral is enabled. In the `initialize()` function it is [done correctly](https://github.com/modm-io/modm/blob/2c29b6780d9d5d56811e148c64e8727b79008ecf/src/modm/platform/spi/stm32/spi_hal_impl.hpp.in#L40).

When I'm doing
```
SPI2->CR1 &= ~SPI_CR1_SPE;
SpiMaster::setDataMode(SpiMaster::DataMode::Mode3);
SPI2->CR1 |= SPI_CR1_SPE;
```

everything works as it should:
[image]

I haven't looked at a solution in detail, but probably the `setDataMode()` function should take care of disabling the peripheral while changing the setting.

hshose, 2023-04-06:
The solution by @chris-durand works well. In addition, we had to add `__DSB()`.
This is what a complete lambda inside a configuration handler could look like:
```cpp
	this->attachConfigurationHandler([]() {
		SPI2->CR1 &= ~SPI_CR1_SPE;
		__DSB();
		SpiMaster::setDataMode(SpiMaster::DataMode::Mode3);
		SPI2->CR1 |= SPI_CR1_SPE;
		__DSB();
	});
```

PR #994 (2023-04-08): [stm32] Fix SPI setDataMode()
Fixes #984 

@hshose
closed 2023-04-08 by rleh