modm-io/modm #1002: STM32F469I-DISCO display does not work on newer board revision (B-08)

hshose, 2023-04-16:
I have a new STM32F469-DISCO board with marking `DK32F469I$AU1` which is supposedly corresponding to a mainboard revision `MB1189-F469NIH6-B08`, according to STs UM1932 manual: 

[image]

The Board has a a new touch panel and "severl part references updated":
[image]

**When I try to flashing any of the display examples, the display doesn't show anything.**

There is a limited datasheet for the FRD400B25025 display in [this thread](https://community.st.com/s/question/0D53W00001pwgZkSAI/stm32f769idisco-breaking-changes-to-touch-hardwaredriver-with-a09-revision), but for now I didn't find helpful information in there to get the display to work.

The people in [this thread](https://community.st.com/s/question/0D53W00000lzPS0SAM/stm32f469idisco-new-lcd-driver-missing-nt35510) seem to have a similar issue and got the `ST TouchGFX Desginer 4.21` as response from ST... not sure how easy it is to extract values from that software. However, the thread says, the new driver might be a *NT35510* instead of the old otm8009a.

For the NT35510, there is a [pretty long datasheet](http://www.lcdwiki.com/res/MRB3973/NT35510_Preliminary_V0.80.pdf) available...
AND... there is an [ST Driver for it on github](https://github.com/STMicroelectronics/stm32-nt35510)

I also found CubeMX generated Code that is supposedly running this driver.
[The `USE_STM32469I_DISCO_REVC` seems to select the version `B08` on the boards marked `DK32F469I$AU1`](https://github.com/mdiepart/lv_port_stm32f469_disco_cubemx/blob/main/Drivers/BSP/STM32469I-Discovery/stm32469i_discovery_lcd.c#L253).

rleh, 2023-06-27:
Let's see the other DSI status registers, `DSI->GPSR`: `0x15`

LOL 🤣 
[image]

hshose, 2026-02-26:
Fixed by #1319