modm-io/modm #1024: monochrome graphic display won't compile - vector ?

lukh, 2023-05-18:
I am really sorry to bring that that way, tried to rebuild an old firmware that used protothread, break because of gcc 13, updating modm for use with fiber...

But something else came: ... I don't know what's happening, and where (and I am not sure I have the required skills to get the problem in the timespawn I have)
Still digging tho, but any help is welcome :)

BTW: Wanted to share how I like modm, you are all doing an amazing job.

Best

`In file included from modm/src/modm/ui/display/monochrome_graphic_display.hpp:83,
                 from modm/src/modm/ui/display/monochrome_graphic_display_vertical.hpp:21,
                 from modm/src/modm/driver/display/ssd1306.hpp:20,
                 from src/osypu_controller.hpp:38,
                 from src/main.cpp:13:
modm/src/modm/ui/display/monochrome_graphic_display_impl.hpp: In instantiation of 'void modm::MonochromeGraphicDisplay<Width, Height, BufferWidth, BufferHeight>::clear() [with short int Width = 128; short int Height = 32; unsigned int BufferWidth = 128; unsigned int BufferHeight = 4]':
src/osypu_controller.hpp:414:34:   required from here
modm/src/modm/ui/display/monochrome_graphic_display_impl.hpp:26:22: error: ambiguous overload for 'operator=' (operand types are 'modm::glcd::Point' {aka 'modm::Vector<short int, 2>'} and '<brace-enclosed initializer list>')
   26 |         this->cursor = {0, 0};
      |         ~~~~~~~~~~~~~^~~~~~~~
In file included from modm/src/modm/math/geometry/vector.hpp:147,
                 from modm/src/modm/math/geometry/circle_2d.hpp:19,
                 from modm/src/modm/math/geometry.hpp:18,
                 from modm/src/modm/ui/display/graphic_display.hpp:25,
                 from modm/src/modm/ui/display/monochrome_graphic_display.hpp:21:
modm/src/modm/math/geometry/vector2.hpp:233:25: note: candidate: 'modm::Vector<T, 2>& modm::Vector<T, 2>::operator=(const modm::Matrix<T, 2, 1>&) [with T = short int]'
  233 |                 Vector& operator = (const Matrix<T, 2, 1> &rhs);
      |                         ^~~~~~~~
modm/src/modm/math/geometry/vector2.hpp:56:15: note: candidate: 'constexpr modm::Vector<short int, 2>& modm::Vector<short int, 2>::operator=(const modm::Vector<short int, 2>&)'
   56 |         class Vector<T, 2>
      |               ^~~~~~~~~~~~
modm/src/modm/math/geometry/vector2.hpp:56:15: note: candidate: 'constexpr modm::Vector<short int, 2>& modm::Vector<short int, 2>::operator=(modm::Vector<short int, 2>&&)'
scons: *** [build/scons-release/src/main.o] Error 1
scons: building terminated because of errors.
`

salkinium, 2023-05-18:
> BTW: Wanted to share how I like modm, you are all doing an amazing job.

Awwww, thanks!

Some debugging ideas in order:
- Try explicitly constructing the cursor: `this->cursor = modm::glcd::Point(0, 0);`.
- Could you try compiling with GCC12? We've had other issues with GCC13 before. 
- Try before C++23 was enabled ([this is the last commit before](https://github.com/modm-io/modm/commit/389a9c352e9d20c1f8ee8a40d5276c2fd3df0557)).
- Finally, the display code was refactored for the 2021q1 release (in #604). Perhaps it was broken since then…

chris-durand, 2023-05-18:
#1004 is probably the culprit. The constructor for `modm::Matrix` is now explicit and previously it worked through some implicit conversions. Apparently, this was not caught by the CI since there is no test for it and the code is not used in any example. 
`this->cursor = modm::glcd::Point{0, 0};` will likely fix it.

lukh, 2023-05-18:
Thanks, I was able to "make it work" by simply revert the matrix related files.
Gonna let it like that for now, and fix it whenever my brain works again ^^

Thanks a lot

lukh, 2023-05-26:
opening a PR #1029

closed 2023-05-27 by salkinium