modm-io/modm #1033: SAMD51/SAME5x cache not enabled?

tarush53, 2023-06-02:
I recently submitted a PR for a FeatherM4 BSP and several examples, one of which was a rudimentary driver for the onboard Neopixel (WS2812) utilizing delay_ns(). Niklas Hauser expressed some interest in that, suggesting that it may be adaptable to the rp2040. I was a little skeptical of the driver being usable as is, because a few instructions inserted by the compiler for a different processor could mess up the timing.

I wish I had a scope to analyze the waveform output, but I don't; so I decided to try measuring the time required to send one 24-bit color to the device by using PreciseClock::now() at the beginning and end of the loop. It seemed as if the extra time required for the instructions outside of the delay_ns() calls were about what they would be for a 120MHz clock with 6 wait states. But I figured the cache must be on, so I kept trying various values for delay_ns() to compensate for the extra time required for the setup instructions of the loop.

It then occurred to me to search modm's initialization code to determine where, exactly, the cache controller was enabled (it is disabled by default). Not finding it, I enabled it in my main.cpp file for testing; and the loop time dropped significantly. I had to conclude that the cache was never enabled at all.

It is easy enough to enable the cache controller in the board.hpp initialization() function, but I would think it should be done in one of the lower level startup files. And using delay_ns() so heavily for this driver, I wonder how that is affected by the cache being disabled?

Thanks,
Tom Rush

chris-durand, 2023-06-02:
Hi Tom, the only reason is that the SAME5x support was never really finished. I had planned to use the chips in a project at work to replace some unobtainable STM32 but they also went out of stock back then. Feel free to open a PR to enable the data cache, if you like.

tarush53, 2023-06-03:
I'm going to answer my own questions :-)

2 and 4. When (if) my Feather M4 BSP PR gets approved and merged, I believe I will submit a PR for 'board.hpp' that turns on the cache in the 'initialization()' function. I don't think it should go in core:cortex-m because the cache was not an ARM option in the M4 boards; it was an add-on done by some vendors and thus vendor specific. Any boards with the SAMD51/SAME5x could also add something to their initialization if desired; but if not, it should probably be mentioned in the docs that the cache is not enabled by default.
At some point it may be useful to have a platform/cache/(vendor) module to handle enabling/disabling, invalidating entries, locking, and other assorted functions of a cache controller.

3. Having the cache enabled does NOT affect the delay_ns() function. I did a test with interrupts disabled, and reading the SysTick value immediately before and after the delay_ns() wrapper call (four 16-bit thumb instructions). Delaying for 50000ns, the result should have been ~6000 clock ticks: with cache off, it was 6067; cache on, 6064; and those numbers were consistent, loop after loop. I interpret that to mean that delay_ns() was running about 1% long with the tuning parameters in the sam platform:core module.lb, and that cache is not faster than modm_fastcode in RAM.

1. A corollary to the previous item is that using the cache memory as TCM with a linker script entry would not be worthwhile.

closed 2023-06-03 by tarush53

chris-durand, 2023-06-03:
To 4: If you want to enable the cache we could put that code into the platform-specific core module, e.g [here](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/sam/startup_platform.c.in).

Enabling the cache should be unproblematic for D51/E5x because we don't support any feature yet that could interfere with caching, like DMA.

tarush53, 2023-06-03:
If you think that wouldn't be too early, then I'll try it there. I just wasn't sure what effect that might have on initializing the memory. (The opening comment in that file sounds so ominous! :-)