modm-io/modm #1035: STM32G0B1 currently black listed

klsc-zeat, 2023-06-05:
The G0 B/C line of devices are currently blacklisted in the repository because of the problem with the shared interrupt lines.
My question would be: What is an acceptable way to support this devices?

1. Quick and dirty patching of the header to have the "old/common" names of the ISR routines as alias.
Pro: easily accomplished
Cons: only 1 of the multiplexed peripherals can run at the same e.g. either I2C2 or I2C3 can be used not both.
2. add support for multiplexed ISRs to I2C, SPI and CAN
Pro: all of the hardware is supported
Cons: quiet complex, lot of work.

So basically what i want to know if a patch for Option 1 has a chance of being accepted or would you need Option 2?

salkinium, 2023-06-05:
We should probably go directly for Option 2. The mechanism of knowing which interrupt needs what peripheral instance callback is already there (via the selected lbuild modules) and we "only" need to convert the individual interrupt handlers into function and call them from a central place. I'm happy to guide you via PR reviews.

chris-durand, 2023-06-05:
I have implemented the shared interrupt handling for the I2C driver and opened a draft PR. Everything is untested since I don't have access to any STM32 hardware right now. SPI should be easy to do because we don't actually use the SPI interrupt handlers in the driver, only DMA handlers for the DMA SPI.

closed 2023-06-07 by salkinium