modm-io/modm #1072: vscode json generation

thestumbler, 2023-09-07:
I noticed a few quirks with the VSCode JSON files generated in the `.vscode` directory using the `build build -m ::vscode` command as [described here](https://modm.io/reference/module/modm-ide-vscode/).

(1) Using these JSON scripts, VSCode reports a deprecated entry:
```
WAS:   "runToMain": true
IS:    "runToEntryPoint": "main"
```
(2) The script generating these JSON files is generating them as JSONC (JSON with comments). Looks like VSCode doesn't have any problem reading it, but my editor NEOVIM was a sea of bright red denoting the JSON "errors"

(3) Documentation search oddness... if you search for VSCODE, there are two results. But if you search for "VISUAL STUDIO CODE", only one result (the page about using VSCode along with WSL2 is missing).

salkinium, 2023-09-07:
I think that code generation module is a little out-of-date by now, we should upgrade it.

thestumbler, 2023-09-07:
One more small thing I noticed yesterday. The very last line (just a right curly bracket) of the automatically generated files doesn't have a newline. It didn't cause any problems that I'm aware of. But the JSON specification may have something to say about that.

closed 2026-10-30 by salkinium