modm-io/modm #1078: Docs Page, Search Function is Intermittent?

thestumbler, 2023-09-13:
Maybe it's something on my side, but I can rarely get the search function to work on the docs page. By work I mean return any results. It does rarely pop up a result, and at first I thought maybe the search was restricted to whatever "chapter" I'm in at the moment. But that doesn't seem to be true. 

Well, just now as I'm trying to find an example, it's suddenly working again. And the scope of the search doesn't appear to matter where you are in the document. So, this is not a bug, unless it comes back again.

rleh, 2023-09-13:
> the docs page

Are you referring to https://modm.io or https://docs.modm.io?

thestumbler, 2023-09-13:
Ah, sorry, I wasn't clear. The latter, docs.modm.io

thestumbler, 2023-09-13:
Maybe so, that sounds typical for my luck. 8-)
It I notice it again, I'll shoot a quick message.

closed 2023-09-13 by rleh

salkinium, 2023-09-13:
The search is completely client-side, so if you have an adblocker or something that blocks Javascript, that may interfere?

thestumbler, 2023-09-13:
Ad blocker, I suppose that's possible. I'm running Safari most of the time but don't take any extraordinary anti-ad steps (to my recollection). I will check this however. I jump between two or more computers and maybe that could be the culprit. 

In the meantime, I keep a couple of TMUX windows open dedicated to the base MODM and the project-generated MODM. That, and a lot of FIND and GREP commands do the trick. In VSCODE I can easily jump around in the generated MODM.