modm-io/modm #1092: Error when trying to compile example on M1 mac

botoxparty, 2023-10-16:
Hi,

I just came across the modm project and it looks perfect for my use case. I'm trying to familiarize myself with the library by building one of the examples for my machine (M1 Macbook). 

I am trying to build and run the `linux/build_info` example. I have set the target to `hosted-darwin` in the project.xml file, and I have also added this to my Sconstruct file `env.Append(CXXFLAGS="-std=c++2b")` so that it uses my installed version of cpp.

However i get this error when trying to run `scons build`:

```
modm/src/modm/platform/core/assert.cpp:29:1: error: argument to 'section' attribute is not valid for this target: mach-o section specifier requires a segment and section separated by a comma
MODM_ASSERTION_HANDLER(_modm_empty_assertion_handler);
```

Any idea how i can get past this?

salkinium, 2023-10-17:
Hm, interesting. On my MBA M2 without modifications:

```
 $ cd modm/examples/linux/build_info
 $ lbuild -D :target=hosted-darwin build && scons 
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
╭────Jinja2──── modm/modm_tools/info.c.in
╰───Template──> modm/src/info_build.c
Compiling C++·· release/main.o
Compiling C++·· release/modm/src/modm/debug/logger/hosted/default_style.o
Compiling C++·· release/modm/src/modm/driver/io/terminal.o
Compiling C++·· release/modm/src/modm/io/iostream.o
Compiling C++·· release/modm/src/modm/io/iostream_printf.o
Compiling C++·· release/modm/src/modm/math/utils/bit_operation.o
Compiling C++·· release/modm/src/modm/math/utils/pc/operator.o
Compiling C++·· release/modm/src/modm/platform/core/assert.o
Compiling C++·· release/modm/src/modm/platform/core/memory.o
Compiling C···· release/modm/ext/printf/printf.o
Compiling C···· release/modm/src/info_build.o
Archiving······ release/modm/libmodm.a
Indexing······· release/modm/libmodm.a
Linking········ /Users/niklas/development/modm-dev/modm/build/linux/build_info/scons-release/build_info.elf
Info:    Project:  build_info
Machine:  blacky
User:     niklas
Os:       macOS 13.5.2 (arm64)
Compiler: g++-12 (Homebrew GCC 12.2.0) 12.2.0
scons: done building targets.
```

Adding `env.Append(CXXFLAGS="-std=c++2b")` to SConstruct works too.

Are you sure you're using Homebrew GCC and not LLVM? I definitely remember linker issues when trying to use the macOS native LLVM.

botoxparty, 2023-10-17:
Ok it was definitely something to do with my GCC/Python.

I ended up uninstalling conda, python and gcc, and then reinstalling python and gcc.
I removed any paths that weren't needed from my .zshrc
I reinstalled xcode-select
I installed gcc with MacPorts

Not sure what fixed it but now it works 😆

closed 2023-10-17 by botoxparty