modm-io/modm #1103: Building Minor Issues on Windows

thestumbler, 2023-12-04:
Just wanted to report this in case anyone else encounters it, but I admit it is an edge case. I was doing some testing on a virtual Windows 11 ARM machine in Parallels on MacOS M1. My project is in a directory that's shared with the host, and mounted as the F: drive. Python is installed (via PYENV) on the main C: drive. 

If you try to LBUILD in this situation, you get an error:

```
F:\projs\hybo\boot23\hello>lbuild build
Traceback (most recent call last):
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.9.9\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.9.9\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.9.9\Scripts\lbuild.exe\__main__.py", line 7, in <module>
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.9.9\lib\site-packages\lbuild\main.py", line 543, in main
    output = run(args)
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.9.9\lib\site-packages\lbuild\main.py", line 527, in run
    builder = Builder(cwd=args.cwd, outpath=args.path, config=args.config,
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.9.9\lib\site-packages\lbuild\api.py", line 65, in __init__
    file_config = ConfigNode.from_file(config)
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.9.9\lib\site-packages\lbuild\config.py", line 191, in from_file
    config.filename = os.path.relpath(filename)
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.9.9\lib\ntpath.py", line 703, in relpath
    raise ValueError("path is on mount %r, start on mount %r" % (
ValueError: path is on mount '\\\\Mac\\files', start on mount 'F:'

```

This seems to be a known limitation of dealing with relative paths on Windows, where there exists no relative path between drives (that is, there is no root folder). I have no workaround for this, so I just temporarily copied the whole project to the C: drive to test further. 

Which led me to this next issue. It might be a Windows 11-only issue, because I'm pretty sure this doesn't happen on a Windows 10 machine (to be confirmed later). When building with Python 3.10.10 active, SCONS fails with this error:

```
C:\files\projs\hybo\boot23\hello>scons build
Traceback (most recent call last):
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\Scripts\scons.exe\__main__.py", line 4, in <module>
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\site-packages\SCons\Script\__init__.py", line 85, in <module>
    from . import Main
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\site-packages\SCons\Script\Main.py", line 61, in <module>
    import SCons.Script.Interactive
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\site-packages\SCons\Script\Interactive.py", line 92, in <module>
    import readline
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\site-packages\readline.py", line 34, in <module>
    rl = Readline()
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\site-packages\pyreadline\rlmain.py", line 422, in __init__
    BaseReadline.__init__(self)
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\site-packages\pyreadline\rlmain.py", line 62, in __init__
    mode.init_editing_mode(None)
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\site-packages\pyreadline\modes\emacs.py", line 633, in init_editing_mode
    self._bind_key('space',       self.self_insert)
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\site-packages\pyreadline\modes\basemode.py", line 162, in _bind_key
    if not callable(func):
  File "C:\Users\rclott\.pyenv\pyenv-win\versions\3.10.10\lib\site-packages\pyreadline\py3k_compat.py", line 8, in callable
    return isinstance(x, collections.Callable)
AttributeError: module 'collections' has no attribute 'Callable'

```
This error goes away if you drop down to Python 3.9.9 or move up to Python 3.11.6

salkinium, 2023-12-04:
Hm, thanks for reporting, but I don't think we can fix this in any meaningful way… I'll keep this open, in case someone else has the same issues.

RavenRandomz, 2025-12-06:
Could this be a bug pertaining to Windows itself? Just a hunch (or my bias towards Linux)