modm-io/modm #1151: [STM32G4] flash dual bank ignored during erase

hshose, 2024-03-20:
Problem: When writing to flash pages beyond `128` on STM32G4 (in particular G474), `Flash::erase` fails silently und the `Flash::program` does not return an error (even though, right after programming the `PROGERR` Bit 3 in `FLASH->SR` register is set, as GDB tells me).

This is due to the `DBANK` bit not beeing set [in the flash.cpp](https://github.com/modm-io/modm/blob/develop/src/modm/platform/flash/stm32/flash.cpp.in#L94)

According to the [RM0440, pp. 135](https://www.st.com/resource/en/reference_manual/rm0440-stm32g4-series-advanced-armbased-32bit-mcus-stmicroelectronics.pdf#page=135), in dual bank mode (`DBANK=1`), the `BKER` bit should be set to `1` for pages after `128`.

What really threw me off was, that I could program addresses beyond page `128` once.
I just now realized, that this this is likely because the factory values for flash would be `0x00` or `0xFF`, in which case they are already erased the first time programmed.

At least the [G474 Flash example](https://github.com/modm-io/modm/blob/develop/examples/nucleo_g474re/flash/main.cpp#L49) and #1146 are effected by this problem.

hshose, 2024-03-20:
Proposed fix in #1146

hshose, 2024-03-20:
LOL might need a bit more logic to model the dual bank flash on the G4 family:
[image]

salkinium, 2024-06-30:
Fixed in https://github.com/modm-io/modm/pull/1146.

closed 2024-06-30 by salkinium