modm-io/modm #1160: Linker issues with newlib version 4.4.0

hshose, 2024-05-03:
After upgrading `arm-none-eabi-newlib` from `4.3.0` => `4.4.0`, I encounter a lot of linker issues, e.g., when building the example in #1146 

<details>
<summary>`getentropy is not implemented`</summary>
```
Linking········ /home/hose/projects/modm_projects/modm/build/nucleo_g474re/flash_json/scons-release/flash_json.elf
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-getentropyr.o): in function `_getentropy_r':
/build/arm-none-eabi-newlib/src/build-nano/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib-4.4.0.20231231/newlib/libc/reent/getentropyr.c:48:(.text._getentropy_r+0xe): warning: _getentropy is not implemented and will always fail
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-getentropyr.o): note: the message above does not take linker garbage collection into account
collect2: error: ld returned 1 exit status
```
</details>

Also, there is a linker error related to `multiple definitions of sprintf`

For now, downgrading `arm-none-eabi-newlib` to `4.3` keeps things working...

rleh, 2024-05-03:
Is the output above shortened? The text says *"warning: _getentropy is not implemented and will always fail"*, so only a warning and the first actual error in your trace is _"error: ld returned 1 exit status"_. Weird.

Anyways, there is no support in modm for GCC 13 (and associated linker and lib versions) yet (see #1012), so this bug will probably not get fixed immediately.

salkinium, 2024-05-03:
Maybe getentropy needs to be stubbed?
https://github.com/modm-io/modm/blob/develop/ext/gcc/cabi_cortex.c#L35

But in general it's best not to mix newlib and gcc versions. The official ARM toolchains have a compatible new,ib version.

hshose, 2024-05-03:
Yeah, no problem that there is no "immediate" fix. I wanted to document this anyways in case other people run into this issue. 

I used to have other weird problems with the gcc 12 that my system shipped... and besides the protothreads, gcc 13 works okay. 

Thanks for the quick replies 🥰🥰

brinzebuah, 2024-11-19:
My solution was to add a modm_config_printf.h file in my workspace to switch of the printf-Stuff
```
#ifndef PRINTF_ALIAS_STANDARD_FUNCTION_NAMES
#define PRINTF_ALIAS_STANDARD_FUNCTION_NAMES 0
#endif``
```
I also removed  "--specs=nosys.specs" from the LINKFLAGS in my SConstruct after read the modm SConscript, because i got the warning: 

> _open is not implemented and will always fail

My question is, why we need the "--specs=nosys.specs" Linkflag at all?