modm-io/modm #1169: fiber/mutex.hpp ::std::scoped_lock is not defined in my environment

TomSaw, 2024-06-16:
Using `modm:processing:fibers`, i get this compiling error:
```
modm/src/modm/processing/fiber/mutex.hpp:192:14: error: 'scoped_lock' has not been declared in 'std'
  192 | using ::std::scoped_lock;
```

Looking into `/usr/arm-none-eabi/include/c++/14.1.0/mutex` i found that it's definition is conditional on `__cpp_lib_scoped_lock`, which is not defined. 
```
ifdef __cpp_lib_scoped_lock // C++ >= 17 && hosted && gthread
  /** @brief A scoped lock type for multiple lockable objects.
   *
   * A scoped_lock controls mutex ownership within a scope, releasing
   * ownership in the destructor.
   *
   * @headerfile mutex
   * @since C++17
   */
  template<typename... _MutexTypes>
    class scoped_lock

...
```

As a workaround, i've branched for `__cpp_lib_scoped_lock`  in `modm/src/modm/processing/fiber/mutex.hpp` too:
```
#ifdef __cpp_lib_scoped_lock
using ::std::scoped_lock;
#endif
```

Here's my output of `arm-none-eabi-c++ -v`, using Fedora 40:
```
Using built-in specs.
COLLECT_GCC=arm-none-eabi-c++
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/arm-none-eabi/14.1.0/lto-wrapper
Target: arm-none-eabi
Configured with: ../gcc-14.1.0/configure --prefix=/usr --mandir=/usr/share/man --with-pkgversion='Fedora 14.1.0-1.fc40' --with-bugurl=https://bugzilla.redhat.com/ --infodir=/usr/share/info --target=arm-none-eabi --enable-interwork --enable-multilib --with-python-dir=share/arm-none-eabi/gcc-14.1.0/python --with-multilib-list=rmprofile --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --enable-languages=c,c++ --with-newlib --disable-nls --disable-shared --disable-threads --with-gnu-as --with-gnu-ld --with-gmp --with-mpfr --with-mpc --with-headers=yes --with-system-zlib --with-sysroot=/usr/arm-none-eabi
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.1.0 (Fedora 14.1.0-1.fc40)
```

salkinium, 2024-06-16:
Hm, looks like it depends on gthread support from the commend `// C++ >= 17 && hosted && gthread`.
Looking at the source, this seems to indeed be the case:
https://github.com/gcc-mirror/gcc/blob/0240909cb03f2a37a74364b00e51ad782c748551/libstdc%2B%2B-v3/include/bits/version.h#L741-L748

That wasn't the case for the stdlibc++ shipped with gcc 12, so I guess I need to ~~copy~~ reimplement that functionality… ugh.

(GCC13/GCC14 is not tested with modm btw, due to the issue with the protothread/resumable macros, see #1012)

salkinium, 2024-06-16:
Actually, can you `#define __cpp_lib_scoped_lock` *before* `#include <mutex>` in the `fiber/mutex.hpp` file? That should enable it I think.

closed 2024-07-10 by salkinium