modm-io/modm #1171: stm32f723 related issues

jeandet, 2024-06-25:
Few issues related to stm32f723 (maybe others f7) 
- it has a single precision FPU while modm builds with `-mfpu=fpv5-d16` instead of `-mfpu=fpv5-sp-d16`
- the integrated USB HS phy seems not totally supported/initialized 
- USB D+ and D- in USB HS are initialized with ` Gpio::OutputSpeed::High` instead of ` Gpio::OutputSpeed::VeryHigh`

I'll try to PR fixes soon.

salkinium, 2024-06-25:
1. Single vs dual precision in F7 is [generated here originally](https://github.com/modm-io/modm-devices/blob/develop/tools/generator/dfg/stm32/stm_device_tree.py#L80-L81).
2. Yup, I didn't find a F723 device with the built-in USB HS PHY in my possession, so couldn't test it.
3. Huh, it's probably also wrong for USB FS?

jeandet, 2024-06-26:
Thx, for the information, I'm actually working on a custom board, I just ordered an stm32f723 discovery board so I'll be able to PR some examples too.
For 1, I think I should spend some time checking which models have double precision FPU and see if there is any rule from the reference. 
I'm also considering making a PR for adding meson build since I'm the QTC meson plugin dev and an hardcore QTC user, would you be open for this too?

salkinium, 2024-06-26:
> I'm also considering making a PR for adding meson build since I'm the QTC meson plugin dev and an hardcore QTC user, would you be open for this too?

YES PLEASE! I like meson, and I'm very happy to help the porting, since I wrote the make and scons one (we don't speak about the cmake one 😒).

niklaut, 2024-07-08:
I've updated the modm-devices submodule to include your changes in this repo, so at least point 1 should be fixed now.

salkinium, 2024-07-08:
I've updated the modm-devices submodule to include your changes in this repo, so at least point 1 should be fixed now.

salkinium, 2025-12-28:
I think all issues here are fixed and the F723 DISCO board is now also supported via TinyUSB.

closed 2025-12-28 by salkinium