modm-io/modm #1182: openocd.py on Windows

minco3, 2024-07-10:
Just started looking into using modm on windows using CMake, building and programming seem to work ok however when I tried to debug using the debug CMake target, there was a python error from `openocd.py`.

```
[build] Traceback (most recent call last):
[build]   File "<frozen runpy>", line 198, in _run_module_as_main
[build]   File "<frozen runpy>", line 88, in _run_code
[build]     call(args.backend(args), args.source, args.config, args.commands, args.ui)
[build]   File "C:\Users\marco\projects\keyboard_firmware\modm\modm_tools\gdb.py", line 48, in call
[build]     with backend.scope():
[build]   File "C:\Users\marco\AppData\Local\Programs\Python\Python312\Lib\contextlib.py", line 144, in __exit__
[build]     next(self.gen)
[build]   File "C:\Users\marco\projects\keyboard_firmware\modm\modm_tools\backend.py", line 35, in scope
[build]     self.stop()
[build]   File "C:\Users\marco\projects\keyboard_firmware\modm\modm_tools\openocd.py", line 70, in stop
[build]     os.waitpid(os.getpgid(self.process.pid), 0)
[build]                ^^^^^^^^^^
```

Which is from here:

https://github.com/modm-io/modm/blob/832f6ce56cf0f7e843e4d937b9106d31a40f194d/tools/modm_tools/openocd.py#L64-L71

Maybe there's a missing indent on line 70?

salkinium, 2024-07-10:
Yes that seems likely. I remember that windows was annoying to get to work and some things are still not implemented properly due to the differences in process management. We should probably switch to PyOCD instead.

Please quickly fix it, you can do it online even.

minco3, 2024-07-10:
created a pull request #1183

closed 2024-07-10 by salkinium