modm-io/modm #1201: CMSIS DSP error when targeting hosted platforms

calebchalmers, 2024-08-19:


salkinium, 2024-08-19:
Ah interesting, I never thought about building CMSIS-DSP for hosted but according to the docs that is supported. Probably also a good idea for testing etc.

Hosted does not generate a `device.hpp` mostly because hosted doesn't have a specific hardware. I would prefer to not include it in the `arm_math.h` for hosted builds. I hope the CMSIS-DSP has sensible defaults for hosted.

salkinium, 2025-12-28:
I'm closing this as there are probably better ways to run CMSIS-DSP on Hosted than via the rather minimal modm support.
There's also an official [Python Wrapper with NumPy support](https://pypi.org/project/cmsisdsp/) which seems more friendly to quick development anyways.

closed 2025-12-28 by salkinium